hackerspace-silesia/Smogomierz #5: Temperatura i wilgotność w API i interfejsie

jozwikp, 2018-12-17:
Brak danych o temperaturze i wilgotności w odpowiedzi API oraz w interfejsie pomimo włączonego czujnika w konfiguracji.

jozwikp, 2018-12-17:
ok widzę w kodzie, że jest funkcja sprawdzająca status czujnika i u mnie pewnie coś nie działa. Sprawdzę jeszcze raz połączenia.

bfaliszek, 2018-12-17:
Jaki masz model czujnika? Jeśli BME280 to powinny być wszystkie dane na stronie /api w takiej formie(JSON) jak poniżej:
``` 
{
  "device_name": "Smogomierz-3909D1",
  "pm1": 102,
  "pm25": 195,
  "pm10": 259,
  "temperature": 4.25,
  "pressure": 1026,
  "humidity": 76,
  "dewpoint": -0.413086
} 
```

jozwikp, 2018-12-17:
Tak cos nie dziala u mnie.
Czujnik z zestawu botland nie dziala.

closed 2018-12-17 by jozwikp

ziyo98, 2018-12-17:
Witajcie, U mnie to samo ( czujnik z zestawu nie działa). Wymieniłem na inny i wszystko śmiga.

jozwikp, 2018-12-17:
czy orientujecie się, czy zadziała DHT11? Jeśli tak to jak podłączyć bo mam tu tylko dout.

bfaliszek, 2018-12-17:
@jozwikp nie testowałem z DHT11, ale działa DHT22 https://github.com/hackerspace-silesia/Smogomierz/blob/master/instrukcje/hardware.md#dht22---miernik-temperatury-i-wilgotności-powietrza Nie polecam używania DHT22 a co dopiero DHT11, one często podają bardzo dziwne dane i skokowo mają jakieś szalone pomiary. Może to nieźle namieszać jeśli włączysz w konfiguracji automatyczną kalibrację pomiarów(wykorzystuje ona wilgotność powietrza do kalibracji pomiarów PM).

@ziyo98 z jakiego modelu miernika temperatury/wilgotności korzystasz? BME280? Część osób, która zamówiła zestawy w Botlandzie dostała wersję Adafriut, podpina się ją troszkę inaczej: https://github.com/hackerspace-silesia/Smogomierz/blob/master/instrukcje/hardware.md#bme280---miernik-temperatury-wilgotności-i-ciśnienia-powietrza Ostatnio wysłali też zestawy z BME280 SparkFun, ale nie miałem okazji ich testować.

ziyo98, 2018-12-18:
dokładnie tego SparkFun'a dostałem: [https://botland.com.pl/pl/czujniki-pogodowe/4901-bme280-cyfrowy-czujnik-wilgotnosci-temperatury-i-cisnienia-atmosferycznego-i2cspi.html](https://botland.com.pl/pl/czujniki-pogodowe/4901-bme280-cyfrowy-czujnik-wilgotnosci-temperatury-i-cisnienia-atmosferycznego-i2cspi.html)  .
On ma jeszcze możliwość adresacji i dołożenia pull-up'a , ale nie miałem czasu tego rozkminiac. Na razie wsadziłem takiego: [https://botland.com.pl/pl/czujniki-cisnienia/11803-bme280-czujnik-wilgotnosci-temperatury-oraz-cisnienia-110kpa-i2cspi-33v.html](https://botland.com.pl/pl/czujniki-cisnienia/11803-bme280-czujnik-wilgotnosci-temperatury-oraz-cisnienia-110kpa-i2cspi-33v.html) .

jozwikp, 2018-12-18:
Ja tez mam tego sparkfuna. ktoś wie już jak go zmusić do działania?

firemark, 2018-12-18:
Czy nie powinno wystarczyć podłączenie SDA, SCL, VCC i GND względem https://github.com/hackerspace-silesia/Smogomierz/blob/master/instrukcje/hardware.md?

Jeżeli jest już dobrze podłączone, to może się okazał że adres do BME280 jest inny niż powinnien być.

Adres możesz zmienić tutaj: https://github.com/hackerspace-silesia/Smogomierz/blob/master/src/bme280.h#L18 - prawd. jest gdzieś w PDFie sparkfuna

bfaliszek, 2018-12-18:
Adafriut domyślnie ma adres 0x77. W Adafriut można podłączyć SDO do GND w ESP8266 i zmienia adres na 0x76. Nie trzeba wtedy zmieniać nic w kodzie. 

W przypadku SparkFun też jest domyślnie 0x77. Chyba w przypadku SparkFun  trzeba zmienić ręcznie w kodzie lub zlutować odpowiednie piny https://learn.sparkfun.com/tutorials/sparkfun-bme280-breakout-hookup-guide/all 

[image]

bfaliszek, 2018-12-18:
@jozwikp zrobię to dziś wieczorem.

bfaliszek, 2018-12-18:
@jozwikp @ziyo98 przygotowałem na szybko .bin dla osób z BME280 SparkFun. Tylko dajcie znać czy działa, bo nie mam BME280 SparkFun i nie wiem czy działa używana przez nas bibliotek. SparkFun ma swoje i mogę je dodać, ale to trochę potrwa. Jeśli będzie ok, to dodam ten .bin do releases. 

https://www.dropbox.com/s/x64tdjx8b40rq9w/Smogomierz_2.4.6_PMS-SparkFunBME280_build_13.05_18.12.2018.bin

jozwikp, 2018-12-18:
404
Tego pliku już tu nie ma

bfaliszek, 2018-12-18:
@jozwikp sorry złego linka wkleiłem. Ten jest poprawny: https://www.dropbox.com/s/12xe23mcz3t5ukp/Smogomierz_2.4.6_PMS-SparkFunBME280_build_13.05_18.12.2018.bin

jozwikp, 2018-12-18:
działa.

bfaliszek, 2018-12-18:
Dodałem ten .bin z obsługą BME280 SparkFun do plików w releases: https://github.com/hackerspace-silesia/Smogomierz/releases

ziyo98, 2018-12-18:
@bfaliszek  - własnie wróciłem z pracy i miałem zabrać się za modyfikację kodu - a tu niespodzianka od Ciebie.  Wielkie dzięki , u mnie też działa.