hackerspace-silesia/Smogomierz #7: Stabilność sprzętu - zawieszony smogomierz

jozwikp, 2018-12-26:
Cześć

Zauważyłem problem ze smogomierzem. Zawiesza się. Rzadko udaje mi się utrzymać go więcej niż dobę w działaniu. Czy ktoś spotkał podobne problemy? Może dobrze byłoby go softem restartować raz na jakiś czas?

ziyo98, 2018-12-26:
Hej,  miałem to samo - pomogła wymiana zasilacza na mocniejszy

bfaliszek, 2018-12-26:
@jozwikp Mam kilka mierników i w żadnym nie mam problemu z wieszaniem się. W jednym czasami przestaje na godzinę lub dwie działać bme280 i podejrzewam, że to kwestia zasilania. Jakich zasilaczy używasz? 
BTW W oprogramowaniu jest systemowy reset ustawiony na 24 godziny ;)

jozwikp, 2018-12-27:
ja mam 5V - 2500mA - za mało?

bfaliszek, 2018-12-27:
2,5A na pewno wystarczy, na 500mA działa bez problemu. Co rozumiesz przez "Zawiesza się"? Nie działa interface www czy nie wysyła danych do zewnętrznych serwisów? Od ponad roku wysyłam pomiary do InfluxDB z kilku Smogomierzy i poza sporadycznymi zawieszeniami się BME280 reszta pomiarów działa bez problemów. 
Możesz też spróbować podłączyć kabel usb bezpośrednio do ESP i włączyć w konfiguracji debug. W monitorze portu szeregowego powinny być komunikaty o każdym pomiarze i ewentualnych błędach.

jozwikp, 2018-12-27:
Nie działa endpoint /api i interfejs www
rusza po restarcie prądowym.

jozwikp, 2018-12-27:
Wersja oprogramowania: 2.4.7 build 13:05_18.12.2018

closed 2019-03-29 by bfaliszek