hackerspace-silesia/Smogomierz #8: Integracja z Firebase

jozwikp, 2018-12-27:
Zrobiłem sobie integrację smogomierza do bazy Firebase. To rozwiązanie jest fajne bo jest za dramo, jest stosunkowo proste i daje dużo możliwości.

Dzięki firebase możecie hostować duże zbiory danych ze smogomierzy i hostować stronę klienta, która te dane potem pokaże. Całość na serwerach google a nie na localhostach.

Klienta buduje się w Java Script i to już mam.

W tej chwili mam pośrednika, który cyklicznie łączy się ze smogomierzem i zasysa dane po API. Chciałbym to pominąć.

Pytanie zasadnicze. Czy na tym module wifi da się uruchomić node.js? Jeśli nie to czy komuś chciałoby się zaimplementować moduł do firebase w c?
https://firebase.google.com/docs/database/cpp/start

lukaszo, 2018-12-27:
Coś takiego da radę https://github.com/FirebaseExtended/firebase-arduino ?

bfaliszek, 2018-12-27:
Jeśli ta biblioteka działa to mogę dodać do Smogomierza. Dobrze, że jest rozwijana, bo widzę ```ArduinoJson is at least version 5.13.1``` i ```ESP8266 Core SDK at least 2.4.1```. ArduinoJson od wersji 6 mocno się zmienił. ESP8266 Core wprowadza sporo zmian w wersji 2.5.0(właśnie wdrażana). Smogomierz używa najnowszych wersji. Jeśli ta biblioteka z nimi działa, to implementacja Firebase, zajmie pewnie z godzina czasu. Sprawdzę to po Nowym Roku i dam znać.

jozwikp, 2018-12-27:
Nie wiem do której bazy jest to firebase-arduino. Podstawowa to realtime database ale jest też firestore. Zakładam, że to biblioteka do realtime database czyli to co potrzeba.

Jeśli to odpalicie to ja to przetestuje i dam znać.

bfaliszek, 2021-11-06:
Do obsługi Firebase na ESP8266/ESP32 polecam https://github.com/mobizt/Firebase-ESP-Client

closed 2021-11-16 by bfaliszek