hackerspace-silesia/Smogomierz #11: Smogomierz nie prezentuje nic oprócz PM z BME280

opennas, 2019-01-05:


bfaliszek, 2019-01-05:
BME280 nie podaje PM, tylko temperaturę, wilgotność i ciśnienie. Jaki masz model BME280? Adafruit? SparkFun? Taki na fioletowej płytce? Jaka wersja oprogramowania dla Smogomierza?

opennas, 2019-01-05:
Chyba nie zauważyłem, że źle podłączyłem :(

GND z NodeMcu do CSB zamiast GND
3V z NodeMcu do SDO zamiast VCC

Mój pierwszy błąd!
Teraz już podłączyłem dobrze. Myślicie, że zjarałem już BME280 ?

Jeśli sądzicie, że żyje to jak uzyskać ciśnienie i wilgotność z CSB i SDO ?

bfaliszek, 2019-01-05:
Nie używasz CSB i SD0, bo Smogomierz używa I2C do połączenia z BME280 a nie SPI. 

SLC podpinasz do D4, SDA do D3, VCC do 3V, a GND do G.

opennas, 2019-01-05:
Teraz mam tak
VCC - 3V
GND - G
SCL - D4
SDA - D3

niestety w "Pomiary" nie pokazuje nic poza PM1, PM2,5, PM10.
W konfiguracji jest włączony BME280 :(

BME280 padł myślisz i dlatego nic nie pokazuje z temperatury, ciśnienia i wilgotności ?

bfaliszek, 2019-01-05:
Sprawdź jeszcze adres BME280. Możesz to zrobić wybierając w ArduinoIDE Plik>Szkicownik>I2C_Scanner. Powinno być 0x76.

opennas, 2019-01-05:
W programie Arduino w ścieżce Plik>Szkicownik> nie pojawia mi się nic
Nie wiem co zrobić, aby tam było >I2C_Scanner :(

W monitorze portu szeregowego też podaje tylko 3 wartości PM.

bfaliszek, 2019-01-06:
To tutaj jest kod https://playground.arduino.cc/Main/I2cScanner Tylko zmień sobie ```Serial.begin(9600);``` na ```Serial.begin(115200);```. Jeśli wszystko masz dobrze podłączone to powinien wykryć połączenie I2C i wyświetlić adres w monitorze portu szeregowego. BME280 ma zazwyczaj adres 0x76 albo 0x77. W kodzie Smogomierza ustawiony jest 0x76. Jeśli masz 0x77 to nie będzie go widzieć.

opennas, 2019-01-26:
Przepraszam za zwłokę, ale dla pewności zamówiłem drugi komplet częsci, bo nie wiedziałem czy nie jest coś uszkodzone. Skorzystałem z kodu do sprawdzenia I2C, który powyżej zaproponowałeś. Zamieniłem szybkość zgodsnie z rekomendacją. Wynikiem jest dziwne zachowanie, bo skanowanie nie daje rezultatu pozytywnego.
Nie wiem co może być przyczyną tego problemu i proszę o pomoc, bo chwilowo wyczerpały mi się pomysły.

20:16:21.928 -> Scanning...
20:16:21.928 -> Unknown error at address 0x01
20:16:21.928 -> Unknown error at address 0x02
20:16:21.928 -> Unknown error at address 0x03
20:16:21.962 -> Unknown error at address 0x04
20:16:21.962 -> Unknown error at address 0x05
20:16:21.962 -> Unknown error at address 0x06
20:16:21.962 -> Unknown error at address 0x07
20:16:21.962 -> Unknown error at address 0x08
20:16:21.962 -> Unknown error at address 0x09
20:16:21.962 -> Unknown error at address 0x0A
20:16:21.962 -> Unknown error at address 0x0B
20:16:21.962 -> Unknown error at address 0x0C
20:16:21.962 -> Unknown error at address 0x0D
20:16:21.962 -> Unknown error at address 0x0E
20:16:21.962 -> Unknown error at address 0x0F
20:16:21.962 -> Unknown error at address 0x10
20:16:21.995 -> Unknown error at address 0x11
20:16:21.995 -> Unknown error at address 0x12
20:16:21.995 -> Unknown error at address 0x13
20:16:21.995 -> Unknown error at address 0x14
20:16:21.995 -> Unknown error at address 0x15
20:16:21.995 -> Unknown error at address 0x16
20:16:21.995 -> Unknown error at address 0x17
20:16:21.995 -> Unknown error at address 0x18
20:16:21.995 -> Unknown error at address 0x19
20:16:21.995 -> Unknown error at address 0x1A
20:16:21.995 -> Unknown error at address 0x1B
20:16:21.995 -> Unknown error at address 0x1C
20:16:22.029 -> Unknown error at address 0x1D
20:16:22.029 -> Unknown error at address 0x1E
20:16:22.029 -> Unknown error at address 0x1F
20:16:22.029 -> Unknown error at address 0x20
20:16:22.029 -> Unknown error at address 0x21
20:16:22.029 -> Unknown error at address 0x22
20:16:22.029 -> Unknown error at address 0x23
20:16:22.029 -> Unknown error at address 0x24
20:16:22.029 -> Unknown error at address 0x25
20:16:22.029 -> Unknown error at address 0x26
20:16:22.029 -> Unknown error at address 0x27
20:16:22.029 -> Unknown error at address 0x28
20:16:22.063 -> Unknown error at address 0x29
20:16:22.063 -> Unknown error at address 0x2A
20:16:22.063 -> Unknown error at address 0x2B
20:16:22.063 -> Unknown error at address 0x2C
20:16:22.063 -> Unknown error at address 0x2D
20:16:22.063 -> Unknown error at address 0x2E
20:16:22.063 -> Unknown error at address 0x2F
20:16:22.063 -> Unknown error at address 0x30
20:16:22.063 -> Unknown error at address 0x31
20:16:22.063 -> Unknown error at address 0x32
20:16:22.063 -> Unknown error at address 0x33
20:16:22.063 -> Unknown error at address 0x34
20:16:22.063 -> Unknown error at address 0x35
20:16:22.097 -> Unknown error at address 0x36
20:16:22.097 -> Unknown error at address 0x37
20:16:22.097 -> Unknown error at address 0x38
20:16:22.097 -> Unknown error at address 0x39
20:16:22.097 -> Unknown error at address 0x3A
20:16:22.097 -> Unknown error at address 0x3B
20:16:22.097 -> Unknown error at address 0x3C
20:16:22.097 -> Unknown error at address 0x3D
20:16:22.097 -> Unknown error at address 0x3E
20:16:22.097 -> Unknown error at address 0x3F
20:16:22.097 -> Unknown error at address 0x40
20:16:22.097 -> Unknown error at address 0x41
20:16:22.130 -> Unknown error at address 0x42
[... 37 lines cut ...]
20:16:22.233 -> Unknown error at address 0x68
20:16:22.233 -> Unknown error at address 0x69
20:16:22.233 -> Unknown error at address 0x6A
20:16:22.233 -> Unknown error at address 0x6B
20:16:22.233 -> Unknown error at address 0x6C
20:16:22.233 -> Unknown error at address 0x6D
20:16:22.233 -> Unknown error at address 0x6E
20:16:22.233 -> Unknown error at address 0x6F
20:16:22.233 -> Unknown error at address 0x70
20:16:22.233 -> Unknown error at address 0x71
20:16:22.233 -> Unknown error at address 0x72
20:16:22.233 -> Unknown error at address 0x73
20:16:22.233 -> Unknown error at address 0x74
20:16:22.266 -> Unknown error at address 0x75
20:16:22.266 -> Unknown error at address 0x76
20:16:22.266 -> Unknown error at address 0x77
20:16:22.266 -> Unknown error at address 0x78
20:16:22.266 -> Unknown error at address 0x79
20:16:22.266 -> Unknown error at address 0x7A
20:16:22.266 -> Unknown error at address 0x7B
20:16:22.266 -> Unknown error at address 0x7C
20:16:22.266 -> Unknown error at address 0x7D
20:16:22.266 -> Unknown error at address 0x7E
20:16:22.266 -> No I2C devices found
20:16:22.266 ->

Marcin999, 2019-12-05:
Witam,
mam identyczny problem. Poniżej dane:
demos D1 mini
dystrybucja: ESP8266_Smogomierz_2.7.2_PMS-SparkFunBME280_build_4.10.2019.bin
VCC - 5V
GND - G
SCL - D4
SDA - D3
Mój czujnik działa na 5V. Sprawdzone na dystrybucji espeasy - działa bez problemu I2C.
Na rynku dwa rodzaje tych czujników 3V i 5V i podłączając go pod 3V na espeasy miałem te same błędy co opennas.
Niestety wyniki z BME280 nie są widoczne. Będę wdzięczny za wskazówki.

bfaliszek, 2019-12-05:
Wersja `ESP8266_Smogomierz_2.7.2_PMS-SparkFunBME280_build_4.10.2019.bin` różni się od `ESP8266_Smogomierz_2.7.2_PMS_build_4.10.2019.bin` innym adresem BME280 w I2C. Jeśli BME280 nie jest widoczne, to może sprawdź oprogramowanie bez `SparkFunBME280` w nazwie.