hackerspace-silesia/Smogomierz #13: Precyzja czujnika BME280 i wyświetlacz

skubi23t, 2019-01-10:
Hej,

W specyfikacji urządzenia widzę, że precyzja czujnika BME280:
Temperature: +-1°C
Humidity: +-3%
Pressure: +-1Pa

Mój czujnik zapewne źle wyświetla temperature. Obecnie w pokoju mam obok siebie 3 czujniki.

BME280: 21.07 stopni | 1015.73 hPa
Xiaomi Aquara: 20.23 stopnia  | 988.85 hPa
Xiaomi: 20.15 stopnia

Jestem w stanie jakos skorygować ten błąd?

Jak chodzi o czujnik PM2.5 to wyniki są bardzo podobne do oczyszczacza Xiaomi AP 2s.

Druga sprawa to podłączenie wyświetacza do projektu - mogę prosić o wskazówkę jak się za to zabrać?

bfaliszek, 2019-01-10:
Pomiar temperatury jest w granicy błędu, więc nie trzeba z nim nic robić. Smogomierz docelowo umieszcza się w obudowie na zewnątrz(oczywiście można go używać też w pomieszczeniach). Pomiar temperatury nie jest do niczego używany w czasie pracy Smogomierza. Pomiar wilgotności powietrza jest istotny, bo używam go do korekcji pomiarów PM2.5 i PM10. 

Jeśli chodzi o jakiś wyświetlacz, to ta biblioteka powinna działać https://github.com/somhi/ESP_SSD1306

skubi23t, 2019-01-10:
Liczyłem, że pomiar temperatury będzie wiarygodny, a różnica 1 stopnia prawie, to już nie jest wiarygodne zwłaszcza teraz jak mamy na zewnątrz 0 czy 1 czy -1...

Jestem zupełnym noobem jak chodzi o ESP8266 i programowanie tego. Dzięki za link poczytam, ale obawiam się, że już nie wiem jak się za to zabrać x)

skubi23t, 2019-01-10:
Ok to jak chodzi o wilgotność to:

BME280 pokazuje: 28.06%
Xiaomi Aquara:  34.90%
Xiaomi: 35.02%

Więc wygląda mi na to, że zaniża..

bfaliszek, 2019-01-10:
Albo Xiaomi zawyża. Nie jestem wielkim fanem ich mierników, bo są bardzo słabej jakości. Na przykład dopiero w Xiaomi AP 2S dali sensowny laserowy miernik PM2.5 i PM10. Wcześniej nawet na tym oszczędzali. Bardziej wierzyłbym boschowi i ich BME280 niż temu co pokazuje Xiaomi. Przykładem może być pomiar ciśnienia z Xiaomi - 988.85 hPa. To wartość dla 0 m.n.p.m. Do poprawnych pomiarów ciśnienia wymagana jest wysokość nad poziomem morza.

skubi23t, 2019-01-10:
Mam 2Sa i faktycznie czujnik PMS7003 a ten z 2S wskazuje bardzo podobne wyniki - prawie idealnie to samo - stoją w salonie w niewielkiej odległości.

W 2S jest też czujnik wilgotności i temperatury. Xiaomi temperature pokazuje 20 stopni i 34% wilgotności.

Smogomierz w tym samym momencie:
Temperatura: 22.28 °C
Wilgotność: 27.30 %

skubi23t, 2019-01-10:
W pokoju mam jeszcze taki zewnętrzny czujnik wilgotności za kilka dolców i temperatury ten pokazuje odpowiednio: 21 stopni i 31%