hackerspace-silesia/Smogomierz #14: Problem z BMP280

kuba-str, 2019-01-20:
cześć

mam problem z I2C.
Mam dwa czujniki takie jak:
http://prntscr.com/m9mqnc
oraz
http://prntscr.com/m9mqwh

Podłączam je prawidłowo. Według instrukcji. Niestety nie są wykrywane. Wgrałem także szkic z https://playground.arduino.cc/Main/I2cScanner. Skaner nie wykrywa ani jednego, ani drugiego termometru.

Macie jakiś pomysł ?
wersja softu to: Smogomierz_2.4.9_PMS_build_19.05_3.01.2019.bin
Zmiana czujnika z BME na BMP w sofcie nic nie daje.

bfaliszek, 2019-01-20:
Jeśli nie wykrywa ich I2cScanner, to muszą być źle podłączone. Spróbuj odwrotnie podłączyć SCL i SDA.

kuba-str, 2019-01-20:
Na NodeMCU też wykrywa, pokazuje mu 0x76. Jednak musiałem zmienić piny 12 i 14. Na 0 i 2 nie czytało w skanerze. Jednak nadal nie podaje wartości, w debugu pokazuje brak danych.

bfaliszek, 2019-01-20:
Sprawdzę to jutro wieczorem, bo mam gdzieś BMP280 i dam znać.

bfaliszek, 2019-01-21:
Głównie BME280 i jednego SHT1x. BMP280 ma raczej tutaj średnie zastosowanie, bo nie mierzy wilgotności powietrza, która jest potrzeba do automatycznej kalibracji pomiarów PM2.5 oraz PM10. Przy zastosowaniu BMP280 pomiary PM będą bez kalibracji i mogą być zawyżone.

kuba-str, 2019-01-21:
mam i taki i taki, jednak wykrywane są tylko pod pinami 12 i 14, nie podają natomiast pomiarów. Czekam na efekt sprawdzania.

bfaliszek, 2019-01-21:
OK, faktycznie BMP280 nie działa. Przy kolejnej aktualizacji oprogramowania postaram się to naprawić.

kuba-str, 2019-01-21:
szukałem, szukałem aż w końcu znalazłem. Skaner pokazywał że czujnik działa na I2C i jego port 0x76. Szukałem dalej, szukałem problemu w bibliotece. Znalazłem kod https://github.com/finitespace/BME280/blob/master/examples/BME_280_I2C_Test/BME_280_I2C_Test.ino. Oparty jest na innej bibliotece i po wgraniu ukazała się wartość wilgotności, ciśnienia i temperatury. Czy można tą bibliotekę włączyć do smogomierza ?

kuba-str, 2019-01-21:
dobra, już wszystko wiem. Testowałem, testowałem aż doszedłem do tego że w tych płytkach NODEMCU i wemos I2C działa tylko na portach D1 i D2. Co za tym idzie, na żadnym innym nie będzie działać. Teraz, wystarczyło zamienić miejscami piny na których działa czujnik smogu i BME280.  Czujnik smogu na 0,2 a I2C na 5 i 4. Teraz elegancko działa. http://prntscr.com/maaj4f

closed 2019-01-22 by kuba-str