hackerspace-silesia/Smogomierz #15: temperatura i wilgotność w homebridge

straubus, 2019-01-22:
Moi drodzy, świetny projekt, ale czy da się dane z bme280 wyciągnąć do homebridge tak samo jak status o jakości powietrza?
Znalazłem taki plugin, https://github.com/lucacri/homebridge-http-temperature-humidity, ale niestety generuje on json o innym formacie i z wyciągania danych nici.

bfaliszek, 2019-01-23:
Plugin do homebridge jest w moim repo i jest tam już wątek rozbudowy: https://github.com/bfaliszek/homebridge-smogomierz/issues/1
Gdy znajdę trochę czasu to postaram się rozbudować tę wtyczkę.

mientki, 2019-02-13:
Niestety po aktualizacji wtyczka nie działa :(

 

> Feb 13 21:09:28 homebridge homebridge[10273]: [2019-2-13 21:09:28] [Parking Jakość powietrza] Smogomierz is working
> Feb 13 21:09:28 homebridge systemd[1]: homebridge.service: Main process exited, code=exited, status=1/FAILURE
> Feb 13 21:09:28 homebridge systemd[1]: homebridge.service: Unit entered failed state.
> Feb 13 21:09:28 homebridge systemd[1]: homebridge.service: Failed with result 'exit-code'.

`

bfaliszek, 2019-04-09:
Wszystkie sprawy związane z wtyczką homebridge najlepiej omawiać w repo tej wtyczki: https://github.com/bfaliszek/homebridge-smogomierz/issues

closed 2019-04-09 by bfaliszek