hackerspace-silesia/Smogomierz #17: Kalibracja

Qbix2, 2019-01-27:
Czy mogę prosić o wskazanie materiałów źródłowych dla warunków i obliczeń w pm_calibration()?

bfaliszek, 2019-01-27:
To z własnych pomiarów i porównań do profesjonalnej stacji pomiarowej, którą mieliśmy minionej zimy.

wiuwiu, 2019-02-05:
Zauważyłem, że zmienne kalibracyjne zależą od modelu. W kodzie widzę trzy modele: red, white, black. Pytanko skąd mam wiedzieć jaki mam model i czym te modele różnią się od siebie? Zakładam, że rozmiarem obudowy, jeżeli tak, to czy mogę prosić o wymiary dla każdego modelu?

bfaliszek, 2019-02-05:
To nie zależy od rodzaju obudowy, a od miernika temperatury/wilgotności i ciśnienia. Te nazwy red, white, black, to nasz wymysł. Jedna z nich to brak kalibracji, kolejna to automatyczna kalibracja(jeśli mamy miernik, który mierzy wilgotność), a trzecia opcja jest wyłączona. Początkowo miała to być taka stała kalibracyjna dla osób, które nie mierzą w ogóle wilgotności powietrza, ale zrezygnowałem z tego. Nie trzeba tutaj nic wybierać i zmieniać. Po wgraniu oprogramowania w karcie Konfiguracja mamy na dolę do wyboru czy chcemy korzystać z automatycznej kalibracji czy nie. Wybieramy, co nam pasuje i zapisujemy zmianę.

closed 2019-02-26 by bfaliszek