hackerspace-silesia/Smogomierz #21: Mocno zaniżone wartości

skubi23t, 2019-02-19:
Mam problem z mocno zaniżonymi wartościami PM oraz zawyżonymi TEMP

https://thingspeak.com/channels/671870

Chciałbym widzieć jak najbardziej zbliżone wartości do tego co faktycznie dzieje się za oknem w danej godzinie, bo mi się to nie pokrywa z airly chociażby... 

Jakie powinny być ustawienia ( pomiar co 10 min i uśredniaj 10 ostatnich ) ? 

W nocy gdzie norma potrafi być przekroczona 5x u mnie jest luzik x)

CapanoThomas, 2019-02-19:
Zwróć uwagę czy sam czujnik PM w obudowie masz przy samych otworach wlotowych, u mnie był z tym problem i przymocowałem go w środku idealnie przy obudowie. Co do temp. to są dwie opcje, czujnik jest na słońcu albo ogrzewa sie od innych elementów.

skubi23t, 2019-02-22:
https://imgur.com/a/nI4dev4

Tak wygląda rozmieszczenie. Jeżeli zdjęcie mało wyraźne zrobię lepsze.

bfaliszek, 2019-02-24:
Jeśli na Smogomierz świeci Słońce, to temperatury mogą teraz wynosić nawet powyżej 20 stopni. Plus widzę, że ESP8266 jest dosyć blisko miernika temperatury i wilgotności. ESP8266 jest najbardziej nagrzewającym się elementem w Smogomierzu.

skubi23t, 2019-02-25:
Poprzestawiałem nieco jak poniżej:
https://imgur.com/a/M6QrWtV

Zobaczymy jaki będzie efekt. 

Jeżeli macie inne sugestie to śmiało.

bfaliszek, 2019-02-25:
Wygląda ok. Daj znać za kilka dni czy pomiary się poprawiły.

skubi23t, 2019-02-26:
Tak to wygląda aktualnie:

https://imgur.com/a/KBhhg9V

Drugi screenshot to airly, powiedzmy, że im wierzę. 

Problem jest w czujniku temperatury i wilgotności. Nie wiem jak to przeskoczyć.

bfaliszek, 2019-02-26:
Poziom PM2.5 jest identyczny. PM10 się różni, ale tutaj wystarczy nawet 100m, żeby wyniki były inne. Jeśli chodzi o temperaturę, to zawsze będzie zawyżona przez nagrzewanie się ESP8266. Jak rozumiem chciałbyś mieć jak najdokładniejsze pomiary temperatury i wilgotności. Dla mnie ważniejsze są pomiary PM, ale planuję dodać opcję usypiania ESP8266 i tylko wybudzanie w celu wykonania pomiarów i wysłaniu danych do airmonitor itd. Dzięki temu ESP nie będzie mieć włączonego ciągle WiFi i temperatura powinna być znacznie niższa i bliższa prawdziwym wartościom.

mientki, 2019-02-26:
Byłoby super. Tak działa Shelly HT - czujnik temperatury i wilgotności na wifi. Wybudza się tylko wtedy, gdy jest wzrost lub spadek z pomiarów. A tak cały czas jest na standby w celu oszczędzania na baterii.

skubi23t, 2019-02-27:
Tak to wygląda teraz

http://imgur.com/nzy8iF3
http://imgur.com/I4NIcie

Co ile najlepiej mierzyć i średnia z ilu ostatnich ustawić ? 1 i 10 wydaje mi się dużo. Aktualnie mam mierz co 5 i usredniaj z 10

Pomysł z odłączaniem wydaje się być bardzo dobry.

Czy da się jakoś dołożyć antenkę z singla wifi żeby polepszyc połączenie? U mnie to działa na gwizdku, potrafi się rozłączyć :(

mientki, 2019-02-27:
ja ustawiałem na 10, 5 i 3 i jakoś różnicy nie widzę. Nie wiem jak to działa do końca. Na pewno nie widać tego na wykresie.
Co trzeba zrobić, żeby być na mapie airly.eu ? Moje smogomierze są jedynie na mapa.airmonitor.pl.

skubi23t, 2019-02-27:
Ale co 10 , 5 i 3 ? 

Tam są dwa ustawienia. Co ile i z ilu średnia.

Nie jestem na mapie airly, ale skoro pokazuje prawie drugie tyle, to coś mi tu nie gra..

mientki, 2019-02-27:
Uśredniaj wynik co 10, 5, 3... nie widziałem różnicy

bfaliszek, 2019-02-27:
Wykresy są generowane przez airmonitor. One są chyba aktualizowane co 15 minut

skubi23t, 2019-02-27:
Jakie sugerujecie ustawić:

A. Wykonywanie pomiarów co =
B. Uśredniaj wynik z ostatnich =
C. Wysyłaj co =

skubi23t, 2019-03-01:
Drugie pytanie. 

Mam dwa punkty oddalone o około 3,6km w linii prostej. 

Mój i czyjś na PMS5003 i BME280. 

Porównanie: https://i.imgur.com/F1erEYw.png

W tym przypadku chodzi o temperature. Telefon mówi, że jest 9 stopni. Może i jest 10. U mnie jest 15..

mientki, 2019-03-01:
Mam to samo z temperaturą. +4 stopnie czasami zawyża nawet 5. Sprawdzone na 2 identycznych smogomierzach. Przydałaby się możliwość kalibracji lub korekty wskazania o określona wartość.

bfaliszek, 2019-03-01:
Macie możliwość sprawdzenia czy to jest stała wartość, czy może ona się zmienia? Jeśli mielibyście wykres np. z 2-3 dni dla tej zawyżonej i poprawnej temperatury, to byłoby idealnie.

skubi23t, 2019-03-01:
Ja jedynie mogę położyć obok czujnik Xiaomi i zbierać dane z niego , ale trudno je udostępnić bo jeszcze nie postawiłem openhaba

mientki, 2019-03-02:
Spróbuje zebrać dane z kilk czujników w przeciągu kilku dni. Może wysnujemy jakieś wnioski.

bfaliszek, 2019-03-02:
Teraz tak myślę, czy nie lepiej będzie pobierać info o temperaturze z jakiegoś api pogodowego i porównywać z pomiarami z BME280 zamiast ręcznie je modyfikować o jakąś wartość. Jeśli różnica będzie większa niż np. 3 stopnie, to podawać dane z api.

mientki, 2019-03-02:
Ale my chcemy mieć pomiary ze swoich czujników a nie z internetu.

mientki, 2019-03-05:
[image]

ziyo98, 2019-03-05:
Ja mam od początku uzywania smogomierza ten sam efekt - obok siebie 3 czujniki - smogomierz w tym momencie pokazuje 6.10stC, czujnik od pieca CO 3,7stC, a ESP8266+DS18B20 - 4,5StC. O ile DS<->CO ma ciągle różnicę 1 st ( jest OK) o tyle smogomierz  zawsze min 2st róznicy.  Zaznacze, ze juz odsunąłem czujnik od reszty elektroniki w obudowie maksymalnie jak to możliwe.

mientki, 2019-03-08:
Może do konfiguracji dodać możliwość korekcji temperatury o wartość do wpisania +-  ??

bfaliszek, 2019-03-08:
> Może do konfiguracji dodać możliwość korekcji temperatury o wartość do wpisania +- ??

Nie chcę takiego rozwiązania, bo ono nie jest uniwersalne. To tak jakbym dodał opcję zmiany poziomu PM o jakąś wartość. Dlatego myślałem o tym api z jakiegoś serwisu pogodowego i jeśli wartość podawana przez BME280 odbiegałaby o >3 stopnie, to używałbym wartości z api, a nie z BME280.

skubi23t, 2019-03-12:
Zdecydowanie przydało by się rozwiązanie na to jakieś.. nie wiem drugi czujnik poza zamontowany tylko do temp zewnętrznej w jakiejś obudowie małej z 3d ?

bfaliszek, 2019-03-23:
Jak będziecie mieć chwilę, to sprawdźcie jak to wygląda na 2.6.0 z włączonym deepsleep. Wtedy pomiędzy pomiarami ESP8266 jest uśpione i w ogóle się nie grzeje, więc pomiary temperatury nie powinny być tak zawyżone – https://github.com/hackerspace-silesia/Smogomierz/tree/2.6.0

mientki, 2019-03-24:
U mnie na jednym z dwóch urządzeń nie ładuje poprawnie tego ustawienia:

[image]

bfaliszek, 2019-03-25:
ESP8266 ma 80KB pamięci, które mam do dyspozycji. Czasami, gdy np. wykonuje pomiary i uśrednia wyniki, to brakuje mu trochę na zrobienie wszystkich tłumaczeń i zostają takie {} nieprzetłumaczone. Po przeładowaniu strony powinno być lepiej. Jest też druga opcja, która powinna pomóc - ustawienie języka angielskiego

mientki, 2019-03-26:
Włączyłem głęboki sen. Od 10 min urządzenie nie wstaje. Tak ma być?

bfaliszek, 2019-03-26:
Tak jak jest w opisie przy deepsleep. Co tyle minut na ile masz ustawione wysyłanie danych do serwisów zewnętrznych. Plus, żeby deepsleep działało, musisz połączyć kabelkiem pin RST z D0.

skubi23t, 2019-03-27:
Dziś połączyłem RST z D0 i włączyłem deep sleep i zdaje się, że działa. Jednak nie widzę, żeby thingspeak zapisywał i dostawał dane. Jakie wartości powinny zostać ustawione?

bfaliszek, 2019-03-29:
Wczoraj chwilę nad tym posiedziałem i faktycznie przy deepsleep nie wysyła poprawnie danych. W 2.6.2 to poprawię.

mientki, 2019-03-29:
Sprawdziłem na 2.6.1. Nie wysyła również danych przez mqtt.

bfaliszek, 2019-03-29:
Jest już 2.6.2 - https://github.com/hackerspace-silesia/Smogomierz/releases/tag/v2.6.2

bfaliszek, 2019-04-09:
Odnośnie pomiarów temperatury, to były też zawyżane przez samo BME280. Miernik ten domyślnie wykonuje pomiary co 0.5ms. W naszym wypadku zupełnie nie jest to potrzebne. W kolejnej wersji oprogramowania wartość domyślna zostanie zmieniona na 1000ms. Dzięki temu samo BME280 nie będzie się już nagrzewać. Z wstępnych testów widzę, że zmiana ta obniża odczyty temperatury o jakieś 0,7-0,8 stopnia.

mientki, 2019-04-09:
Tego się nie spodziewałem :) sprawdzałeś o ile obniży się temperatura po zwiększeniu czasu nawet do minuty?

skubi23t, 2019-04-09:
Dla mnie to temperaturę mógłby sprawdzać i co 5 minut a dokładnie może zaraz przed tym zanim ruszy ESP i "nagrzeje" wnętrze?

mientki, 2019-04-09:
Akurat temperatura, wilgotność i ciśnienie jie zmieniają się tak szybko wiec odczyt co 1-5 min będzie ok. Trzeba tylko sprawdzic czy będzie różnica.

skubi23t, 2019-04-13:
Coś nie mogę zaktualizować... przesyła 19% i nagle bum odcina smogomierz..

bfaliszek, 2019-04-13:
@skubi23t jakie 19% gdzie masz jakieś procenty?

skubi23t, 2019-04-13:
Browse - wybierasz plik > update i widać po lewej na dole ile % zostało przesłane... 
Ustawiłem aktualizacje automatyczna i przenioslem na chwile smogomierz blizej routera i załapał.

Jest szansa na wstawienie do panelu gdzies % jak chodzi o połączenie z danym WIFI ?

bfaliszek, 2019-04-13:
Jest możliwość dodania tego jaki ma zasięg WiFi w %. W przyszłości dodam gdzieś takie info.

bfaliszek, 2019-04-15:
Wstępnie zrobiłem coś takiego. Ukryję RSSI, bo mało mówi o sygnale, zostawię tylko to % info. Dodam jakieś opisy. Zostawię chyba na stronie Update. Chociaż bardziej pasowałoby w Konfiguracji.
[image]
cc @skubi23t o coś takiego Ci chodziło?

mientki, 2019-04-15:
RSSI mówi bardzo dużo. To strata sygnału w dB. Ja bym to zostawił.

mientki, 2019-04-16:
Jak mocno macie zawyżony pomiar w sofcie 2.6.5 ? U mnie jest nadal spora różnica. Około 2 stopnie.

aktualnie
BME280 na NodeMCU (smogomierz)= 7.87 °C
DS18B20 na sonoff basic = 5.9°C

bfaliszek, 2019-04-16:
Przy podawaniu wartości % użyłem opisanego tutaj https://stackoverflow.com/a/15798024 wzoru

> większe lub rówane -51 dBm a -73 dBm - moc sygnału bardzo dobra
> pomiędzy -75 dBm a -85 dBm - moc sygnału dobra
> pomiędzy -87 dBm a -93 dBm - moc sygnału słaba (poszukać lepszej lokalizacji lub zastosowanie anteny zewnętrznej.)
> mniejsze niż -95 dbm - moc sygnału bardzo słaba

skubi23t, 2019-04-16:
Tak o to mi chodziło. Dzięki. 
Czekam na nowy update, przy czym deep sleep nie do konca mi sie podoba. W koncu po to mam smogomierz w lan, żeby wejsc i sprawdzic - a nie czaić się na okazję kiedy będę mógł podglądnąć obecne wskazania

ookris, 2019-04-16:
@mientki u mnie było podobnie przez jakiś czas. Ale chyba po włączeniu automatycznej kalibracji wszystko wróciło względnie do normy 

@skubi23t deep sleep przydaje się kiedy wysyła się dane do zewnętrznej bazy danych co np 5 minut.

mientki, 2019-04-16:
u mnie deep sleep działa bardzo dziwnie. Ustawie 1 min to się wybudzi kilka razy, dość nieregularnie, później przerwa długa, się czasami wybudza. Nie ma reguły. Może więcej czasu zajmuje mu łączenie z wifi i przerywa cały proces - to tylko domysły :) generalnie deep sleep nie używałbym w ogóle, gdyby czujnik BMP280 nie zawyżałby wartości.

bfaliszek, 2019-04-16:
minuta przy deepsleep nie ma specjalnie sensu, jeśli po wybudzeniu Smogomierz czeka 30 sekund, robi serię pomiarów i się usypia(całość trwa jakieś 1-1,5 minuty). @mientki ustaw deepsleep na 3 minuty i wtedy zobacz jak będzie działał.

CC @skubi23t deepsleep jest raczej dla osób, które nie korzystają z interfejsu WWW. Jeśli masz kilka mierników, lub swoją bazę danych i np. Grafanę, to nie wchodzisz na każdy z mierników, żeby zobaczyć odczyty.

closed 2021-11-16 by bfaliszek