hackerspace-silesia/Smogomierz #23: Wsparcie dla GSM/GPRS

wparuch, 2019-02-21:
Czy jest szansa na rozszerzenie projektu o pakietową transmisję danych? Byłoby to bardzo przydatne w miejscach gdzie nie ma wifi, a jest zanieczyszczone powietrze :). Np przy użyciu modułu SIM5320A / SIM800 / SIM900 / SIM908 ale to bardzo luźne propozycje.

firemark, 2019-02-23:
Cześć - jak jesteś z okolic Śląska to możemy się omówić i sami napisać obsługę i sposób podłączenia do esp :)

wparuch, 2019-02-24:
Niestety, małopolskie

bfaliszek, 2019-02-24:
Ciekawy pomysł. Myślę też o możliwości zapisu pomiarów w pamięci ESP lub na karcie pamięci i wysyłaniu ich ich dopiero, gdy będzie dostęp do sieci lub co kilkadziesiąt minut. 
Jeśli chodzi o transmisję danych GSM/GPRS to nie powinno być większego problemu. Tylko tu chyba też byłoby lepiej wysyłać zbiorczo pomiary raz na kilkadziesiąt minut, niż np. co minutę czy tam pięć.

wparuch, 2019-02-24:
Cache'owanie danych to też fajny pomysł, historia byłaby ciągła.

Aczkolwiek przy GPRSie osobiście wolałbym mieć odczyt +- na żywo, analogiczny jak po WiFi. Można zrobić na to flagę wg preferencji usera.

bfaliszek, 2019-02-24:
Jakiś czas temu, znajomy podrzucił mi pomysł montowania Smogomierza na dronie, dlatego zacząłem się zastanawiać nad jakimś cache'owaniem danych i wysyłaniem ich dopiero przy dostępnie do WiFi. GPRS mógłby pozwolić na dostęp do tych danych na bierząco. Tylko tutaj problemem chyba będzie waga baterii, która musiałaby to wszystko zasilać i jeszcze zmieścić się na dronie ;)

wparuch, 2019-02-24:
Niezłe! Myślę, że na kilkunastominutowe loty (bo na więcej zapewne bateria drona nie pozwoli) ciężar baterii byłby to do przeskoczenia.

wiuwiu, 2019-02-27:
Dron DJI Phantom dałby radę unieść taką konstrukcję. Swego czasu miałem taką konstrukcję Ardunio z modułem GSM i podpiętym czujnikiem pyłów. Całość zasilana z dołączonej baterii. [https://www.dfrobot.com/index.php?route=product/product&product_id=1448&search=dfr0355&description=true&category_id=48#.VzrutOT2HHY](url)

closed 2021-11-16 by bfaliszek