hackerspace-silesia/Smogomierz #24: Problem z wysyłką danych do mapa.airmonitor.pl

bjungiewicz, 2019-02-23:
Witam, urządzenie nie wysyła danych do serwisu, nie wiem w czym problem. W serwisie aitmonitor twierdzą, że jeśli dane geo w urządzeniu i formularzu są identyczne to mapa automatycznie powinna po ok. 3 godzinach wyświetlać informacje. 
Może źle podaję dane? Smogomierz przyjmuje dane do 6 miejsca po przecinku w formacie np. "51,092400" a formularz airmonitor do czwartego miejsca z kropką zamiast przecinka "51.0924". 
Jak zweryfikować czy urządzenie wysyła dane tak jak powinno?

bfaliszek, 2019-02-24:
Z tego co widzę, to na airmonitor trafiają Twoje dane. Z takimi współrzędnymi jak podałeś:
http://metrics.airmonitor.pl:3000/d/000000002/airmonitor?orgId=1&var-latitude=51.092400&var-longitude=All
Smogomierz wysyła dane pomiarowe z zapisanymi współrzędnymi do 6 miejsc po przecinku.

skubi23t, 2019-02-25:
https://www.facebook.com/permalink.php?story_fbid=310301466355496&id=131160177602960

Może potrzebujemy to troszkę zmienić?

skubi23t, 2019-02-26:
Ok okazuje się, że jak uzupełniacie formularz airmonitor to musicie wpisać dokładnie te same dane jak z ustawien smogomierza (jedynie przecinek zamienic na kropke).

Po kilku godzinach pojawia się punkt na mapie.

bfaliszek, 2019-02-26:
Wydawało mi się logiczne, że w formularzu musimy wpisać te same współrzędne co w Smogomierzu. Jak rozumiem wszystko jest już OK i mogę zamknąć wątek.

closed 2019-02-26 by bfaliszek