hackerspace-silesia/Smogomierz #25: Wsparcie dla MQTT i czujnika BME680.

wiuwiu, 2019-02-27:
Z ciekawych rzeczy fajnie by było mieć obsługę MQTT w Smogomierzu by móc przesyłać dane bezpośrednio do lokalnego systemu inteligentnego domu. 
Kolejną ciekawą rzeczą jest obsługa czujnika BME680 z pomiarem VOC / gazów. Bosh udostępnia firmware dla ESP8266 do obliczania indeksu jakości powietrza IAQ. Link: [https://botland.com.pl/pl/czujniki-gazow/10872-adafruit-bme680-czujnik-temperatury-wilgotnosci-cisnienia-i-gazu.html](url)

bfaliszek, 2019-02-27:
Zakładam, że Smogomierze są umieszczane raczej na zewnątrz, więc pomiar IAQ (Indoor air quality), plus to co podaje BME680 to nie jest wartość VOC, tylko rezystancja wszystkich gazów. Myślałem o dodaniu CCS811, który podaje właśnie VOC i CO2, ale on nie podaje wartości absolutnych i każdy z nich wymaga kalibracji, więc raczej za dużo z tym zachodu. Chciałbym dodać coś do pomiaru NO2 ale większość mierników nie radzi sobie z niskimi wartościami, a te które podają wartość od 0, kosztują majątek.

wiuwiu, 2019-02-27:
Wzięte z opisu czujnika: 

> Czujnik BME680 zawiera mały czujnik MOX. Ogrzewany tlenek metalu zmienia rezystancje w oparciu o lotne związki organiczne (VOC) znajdujące się w powietrzu, dzięki czemu może wykrywać gazy i alkohole, takie jak etanol, alkohol i tlenek węgla. Może być także wykorzystywany do wykonywania pomiarów czystości powietrza. 

> 
W przypadku CCS811 można by oprzeć się nie tyle na samych wartościach co na ich zmianie. Wówczas kalibracja nie byłaby potrzeba a każdy by widział, że jeżeli wartości nagle strzelają w górę to "wiedz, że coś się dzieje" :)

bfaliszek, 2019-03-23:
W 2.6.0 dodałem wstępnie wsparcie dla MQTT, tylko nie miałem za bardzo jak przetestować, więc będę wdzięczny na info czy działa – https://github.com/hackerspace-silesia/Smogomierz/tree/2.6.0

mientki, 2019-03-24:
Wygląda na to że coś tam się pojawia w brokerze...

[image]
 Możnaby ewentualnie dodać w składni topiców nazwę urządzenia, np. tak:

Smogomierz-8421EF/sensor/temperature

gdzie "Smogomierz-8421EF" to nazwa urządzenia. Przy dwóch smogomierzach na tym samym brokerze MQTT jest konflikt, a wyniki nadpisują się pod takim samym topikiem.

bfaliszek, 2019-03-24:
@mientki dzięki za info. Dodam to w kolejnej aktualizacji.

bfaliszek, 2019-03-29:
Jest już 2.6.2 - https://github.com/hackerspace-silesia/Smogomierz/releases/tag/v2.6.2 poprawiłem m.in. tam wsparcie dla MQTT

mientki, 2019-04-07:
Mam 2 smogomierze w sieci:
1) Nazwa urządzenia: Smogomierz-8421EF 
2) Nazwa urządzenia: Smogomierz-0A5017 

Na brokerze MQTT pojawiają zupełnie inne nazwy:

[image]

Z czego to może wynikać?

bfaliszek, 2019-04-07:
Nazwa urządzenie jest trochę inaczej generowane. To co jest w brokerze to ten sam ChipID, który jest podany na stronie Konfiguracji przy opcji włączenia luftdaten.

mientki, 2019-04-07:
A jak jest generowany? Możnaby to ujednolicić.

closed 2021-11-16 by bfaliszek