hackerspace-silesia/Smogomierz #26: Sensirion SPS30

CapanoThomas, 2019-03-04:
Cześć

wpadł mi w ręce czujnik PM2.5 PM4 i PM10 Sesiriona SPS30, czy istnieje opcja, żeby go jakoś podłączyć do naszego Smogomierza?

bfaliszek, 2019-03-04:
Bardzo fajny miernik. Polecam 😁

[image]

Co do wsparcia w Smogomierzu to pewnie dodam w przyszłości, ale raczej wątpię, żeby wiele osób go użyło w tym projekcie. SPS30 kosztuje jakieś 160 zł. Za tyle można mieć pełen komplet części na Smogomierz.

CapanoThomas, 2019-03-04:
HEHE, o proszę, a masz może do niego jakiś fragment kodu/szkic pod arduino i NodeMCU żeby zrzucał dane do Gafany? Przyznam, że potrzebuję pomocy w tym temacie.

bfaliszek, 2019-03-04:
@CapanoThomas Mam kod dla ESP32. Nie sprawdzałem czy działa z ESP8266. https://github.com/bfaliszek/SensirionSPS30_InfluxDB

CapanoThomas, 2019-03-04:
@bfaliszek dzięki wielkie, czekam na wtyczkę jeszcze i będę testował. 
BTW, na firmę ten czujnik kosztuje 114zł w SOS, a jak rozumiem wygrywa żywotnością i dokładnością pomiarów.

closed 2019-03-04 by CapanoThomas