hackerspace-silesia/Smogomierz #30: Wyliczanie punktu rosy

skwaruch, 2019-03-12:
Jakie jest źródło wzoru wyliczającego punkt rosy w "smogomierz"?
        message += "<h3>Dew point: ";
        message += (BMESensor.temperature - ((100 - BMESensor.humidity) / 5));
        message += " °C</h3>";

bfaliszek, 2019-03-12:
Tak, to uproszczony wzór do obliczania punku rosy: https://en.wikipedia.org/wiki/Dew_point#Simple_approximation
Nic nie stoi na przeszkodzie, żeby wprowadzić bardziej złożone.

closed 2019-03-12 by bfaliszek

skwaruch, 2019-03-12:
Powiedział bym bardzo bardzo uproszczony.
Dodajcie taki:
dewPoint = pow(BMESensor.humidity/100, 0.125)*(112+0.9*BMESensor.temperature)+0.1*BMESensor.temperature-112;

bfaliszek, 2019-03-12:
OK, poprawiłem sobie w kodzie. Punkt rosy, według podanego wzoru będzie obliczany w wersji oprogramowania 2.5.1.

skwaruch, 2019-03-12:
Fajnie. 
Czy chcecie bym napisał co jeszcze pozmieniałem?

bfaliszek, 2019-03-12:
Jasne. W 2.5.1 dodałem możliwość ustawienia loginu i hasła dla stron z konfiguracją oraz update. Dodałem też obsługę MQTT(ale nie testowałem jeszcze czy działa). Planuję też podzielić plik webserver.h na mniejsze pliki. Ułatwi to tłumaczenie interfejsu oraz umożliwi dodanie różnych skórek dla panelu użytkownika.