hackerspace-silesia/Smogomierz #33: Nowe UI

ookris, 2019-03-24:
Trochę mi zeszło ale pierwsza wersja już jest i jakoś nawet działa. Choć czasami nie wszystkie tłumaczenia ładują się prawidłowo. Czasami zostają 2-3 nieprzetłumaczone opcji w {}. Czasami też strona konfiguracyjna nie chce się załadować w ogóle. 

Zmienione pliki:
https://drive.google.com/file/d/1eovqppQNktdRpIpWhK2qY8hUoBgWV1Hz/view?usp=sharing

bfaliszek, 2019-03-24:
Zostaje kilka {}, bo kończy się pamięć w ESP. Postaram się w wolnej chwili trochę zoptymalizować jej zużycie, żeby coś zaoszczędzić i nie było problemów z tłumaczeniem 😉

mientki, 2019-03-24:
Pokażesz jak wyglada to nowe GUI?

ookris, 2019-03-24:
Całość będzie nieco szersza i jaśniejsza. Kilka rzeczy trzeba jeszcze ogarnąć ale ogólny zarys już jest. 
[image]
[image]

krupniok200, 2019-03-24:
powiem że strona w takim stylu jest dużo czytelniejsza, szczególnie strona z pomiarami, konfiguracja czy update ważne aby działała :)

mientki, 2019-03-24:
Fajnie, że podświetla pomiar kolorem. Konfiguracja nie chce się u mnie załadować, wykresy zrobiłbym wyższe razy 2.

ookris, 2019-03-25:
@mientki problem z ładowaniem strony konfiguracyjnej wynika, jak twierdzi @bfaliszek, z kończącej się pamięci w ESP. Wykresy można zrobić wyższe 2x ale w przypadku kiedy włączymy wyświetlanie wszystkich strona zrobi się nieco dłuższa i żeby dotrzeć do interesujących wykresów trzeba będzie sporo przewijać. Mógłbym porobić zakładki dla każdego z serwisów generujących wykresy ale nie wiem czy ESP, które jest już nieco przeładowane, to wytrzyma.

bfaliszek, 2019-03-25:
ESP8266 ma 80KB pamięci dla zmiennych. Normalnie zajęte jest jakieś 63% z tego. Problem robi się, gdy wjeżdża tłumaczenie. Tam jest sporo Stringów, które zajmują pamięć. Gdy się kończy to zostają nieprzetłumaczone {}. Jak będę miał chwilę, to muszę spróbować zmienić te Stringi na char, żeby zaoszczędzić pamięci.

mientki, 2019-03-25:
Optymalizować :) osobiście wróciłem do oryginału bo u mnie ma po prostu działać zawsze.

mientki, 2019-03-25:
A nie lepiej zrobić oddzielne kompilacje dla różnych języków? Tasmotę właśnie tak wypuszczają.

bfaliszek, 2019-03-25:
Docelowo chcę tak zrobić i po zmianie w ustawieniach pobierałoby odpowiedni .bin z serwera i instalowało. Tylko najpierw chciałem wypuścić to co mam. Dodać automatyczne aktualizacje i później zacząć optymalizować i dzielić całość na mniejsze części. Żeby zrobić całość tak jak chcę, czyli kilkanaście mniejszych .bin w każdej kombinacji miernika pm z temp/humi i językami, muszę mieć działające automatyczne aktualizacje.

ookris, 2019-03-25:
Opcja osobnej kompilacji dla każdego z języków to chyba dobry pomysł. 
@mientki strona z odczytami działa poprawnie bo niewiele jest tam do tłumaczenia.

ookris, 2019-03-28:
Nowa wersja Smogomierz już jest i jest też wersja z moimi modyfikacjami wyglądu. 
Zrezygnowałem jak na razie ze zmiany wyglądu strony konfiguracyjnej oraz strony z aktualizacją FW.
Zmiany mają charakter czysto wizualny także wszystkie pomiary powinny "działać".

closed 2021-11-16 by bfaliszek