hackerspace-silesia/Smogomierz #34: interwał wysyłania

krupniok200, 2019-03-26:
w ostatnich aktualizacjach został ujednolicony interwał wysyłania danych do serwisów zewnętrznych. Dziś się dowiedziałem że Luftdaten ukrywa znacznik jeżeli w ciągu 5min nie dostanie nowych danych. Przy dzisiejszych ustawieniach mija się to z celem bo nie chcę wysyłać co 5min danych do baz danych, a chcę mieć non stop widoczny znacznik u Niemców.

Wyjście z problemu jakie proponuje:

- powrót do osobnych interwałów dla każdego serwisu
- zrobić osobno interwał dla baz danych (thingspeak, influxdb) i osobno dla serwisów (airmonitor, luftdaten)
- zrobić sztywny interwał dla luftdaten na 5min chyba że pomiary wew będą więcej niż co 5min do wyrównanie do tej liczby

bfaliszek, 2019-03-26:
Ujednoliciłem interwały wysyłania danych, bo zależało mi na opcji deepsleep i usypianiu całego esp. Czas uśpienia, to właśnie ten interwał. Jeśli zrobię oddzielne interwały, to i tak jakoś będę musiał wybrać czas uśpienia dla deepsleep.

krupniok200, 2019-03-26:
Każdy czas pomiędzy wysyłaniami, to zawsze jakaś oszczędność energii a ja np sleepa nie korzystam bo mam stałe zasialnie z 230V

bfaliszek, 2019-03-26:
Jakieś 90-95% energii zużywa wifi i ciągle działający webserwer. Deepsleep to usypia, właśnie dlatego chciałem dodać tę opcję. Sam mam zasilanie z sieci, ale wolę robić deepsleep i mieć interwał 5 minut. Dodatkowo ESP8266 się tak nie grzeje i nie zawyża pomiarów temperatury.

krupniok200, 2019-03-27:
To jak zamierzać rozwiązać ten problem?

bfaliszek, 2019-03-27:
Przy Deepsleep będzie 1 interwał do wszystkiego, a bez deepsleep zrobię dwa(jeden dla luftdaten/airmonitor oraz oddzielny dla thingspeak/influxdb).

krupniok200, 2019-03-27:
I tak będzie najlepiej :) wilk syty i owca cała :)

bfaliszek, 2019-03-29:
Dwa interwały przy wyłączonym deepsleep i jeden przy włączonym deepsleep pojawi się w oprogramowaniu 2.6.2

bfaliszek, 2019-03-29:
Jest już 2.6.2 - https://github.com/hackerspace-silesia/Smogomierz/releases/tag/v2.6.2

closed 2019-03-29 by bfaliszek

krupniok200, 2019-03-29:
Nie zapisuje się ustawienie interwału dla baz danych, zmieniam na 30min a po restarcie dalej jest 2min

bfaliszek, 2019-03-29:
Pobierz jeszcze raz 2.6.2 i zaktualizuj Smogomierz. Powinno pomóc :)