hackerspace-silesia/Smogomierz #35: changing chipID may cause collision with other sensors on Luftdaten.info

ricki-z, 2019-03-27:
At the moment Luftdaten.info doesn't accept regsitrations for devices with this firmware.
The possibility to change the chipID may lead to collisions with other sensors (not only for Luftdaten.info).
Please contact me for more infos (we had a collsion with the third registration of such a device).
You could change the device prefix in "X-Sensor" (src/luftdaten.cpp) from esp8266 to smogomierz . Users then need to send "smogomierz-"+chipID as the chipID at registration. So collisions may only affect users of this firmware.

bfaliszek, 2019-03-27:
thanks for the information. I will correct the code as suggested. I will change the X-Sensor to a smogomierz

bfaliszek, 2019-03-27:
1) Removed change ChipID option: https://github.com/hackerspace-silesia/Smogomierz/commit/173f307d755490a2c5fe84594a6c6be9225e7a8c
2) Added "smogomierz" to X-Sensor: https://github.com/hackerspace-silesia/Smogomierz/commit/838135d37aec250721df52e6304eeee744c49d56

ricki-z, 2019-03-27:
could you please also add the prefix to the config page so that user will see what "chipID" they have to use for registration?

bfaliszek, 2019-03-27:
Yes. I also added it to the configuration page https://github.com/hackerspace-silesia/Smogomierz/commit/173f307d755490a2c5fe84594a6c6be9225e7a8c#diff-8f8f75b09c165f6369f2f745e4b7ccefR388

[image]

ricki-z, 2019-03-27:
Then I can close this issue.

closed 2019-03-27 by ricki-z