hackerspace-silesia/Smogomierz #41: Uwagi, propozycje.

skwaruch, 2019-04-15:
Nie działa pomiar dla PMS7003:
1. prawie zawsze są 0 
Ciągłe pomiary PM: nie
Wykonywanie pomiarów PM co: 1minut
Uśredniaj wynik z ostatnich: 1pomiarów PM
Wysyłanie danych do AirMonitor/Luftdaten/Smoglist co: 2 minut
Wysyłanie danych do ThingSpeak/InfluxDB/MQTT co: 2	minut
DeepSleep: nie
2. wartości mają duże wahania, również wartości 0
Ciągłe pomiary PM: nie
Wykonywanie pomiarów PM co: 1minut
Uśredniaj wynik z ostatnich: 1 pomiarów PM
Wysyłanie danych do AirMonitor/Luftdaten/Smoglist co: 2 minut
Wysyłanie danych do ThingSpeak/InfluxDB/MQTT co: 2	minut
DeepSleep tak
3. prawie zawsze są 0
Ciągłe pomiary PM: nie
Wykonywanie pomiarów PM co: 1minut
Uśredniaj wynik z ostatnich: 8 pomiarów PM
Wysyłanie danych do AirMonitor/Luftdaten/Smoglist co: 2 minut
Wysyłanie danych do ThingSpeak/InfluxDB/MQTT co: 2	minut
DeepSleep tak

Jakie powinny być ustawienia by to działało?

Dla akumulatora 18650 (około 2Ah) smogomierz z PMS7003 nie będzie działać dłużej niż dwa (niezależnie od częstotliwości pomiarów) z powodu nie wyłączania PMS7003 (nie wiem co robi pms.sleep(), ale na pewno nie wyłącza wiatraczka, który odpowiada praktycznie za cały pobór prądu czujnika). Czujnik PMS pobiera mniej więcej tyle samo co ESP.
Mój smogomierz działa na takim aku 2-3 dni, ESP załączane co 4 min na około 1 min (30 sekund mniej więcej trwa start drugie 30 sek trwa pomiar), wraz z EPS wyłączane jest PMS wraz z wiatraczkiem.
Powiedzcie dla jakiej baterii prawdziwe jest "możliwość pracy na baterii przez kilka tygodni"? ;)

Co do pracy na baterii to ja sobie dodałem pomiar napięcia za pomocą ADC_MODE(ADC_VCC)
Ja wyliczam sobie średnią z 10 pomiarów w ciągu 30 sekund bo mam straszne wahania (spadki napięcia przy większym poborze prądu) napięcia.

W skrócie ja mam zrobione:
1. pomiar napięcia wraz z wysyłaniem
2. punkt rosy wraz z wysyłaniem
3. wyłączanie PMS
4. zmiana mocy (TX_POWER)
5. proste, optymalne, tanie zasilanie z aku 18650 :)

Jeśli kogoś coś z tego interesuje to proszę pisać, chętnie podzielę się tym co zrobiłem.

bfaliszek, 2019-04-15:
Najwidoczniej nie działa Ci PMS7003. Chyba, że miałeś starsze oprogramowanie (2.6.2 lub wcześniejsze), wtedy podawało przy pierwszym pomiarze wartość 0. Zwłaszcza przy deepsleep.

pms.sleep() właśnie usypia PMS7003 i wiatraczek w nim.

Najprawdopodobniej masz coś pomieszane w kodzie lub masz coś źle podłączone. Wgraj na czystko 2.6.5 i zobacz jak działa deep sleep. Teraz chodzi poprawnie i sam z niego korzystam w 2 Smogomierzach.

skwaruch, 2019-04-15:
Myślę że nie mam źle podłączone (zanim wgrałem wersję 2.6.5 wszystko działało), oczywiście mam czysty 2.6.5.
Jeszcze zauważyłem: smogomierz wysłał wartości 0 a po chwili na stronie pojawiły się poprawne wartości.

U mnie pms.sleep() wyłącza wiatraczek na sekundę (tak jak pms.wakeUp()).

bfaliszek, 2019-04-15:
Przy włączonym deepsleep faktycznie wysyła 0. Po wyłączeniu jest ok. Poprawię to przy następnej aktualizacji.

skwaruch, 2019-04-15:
https://thingspeak.com/channels/629989
Tak od godziny 14:20 są parametry ja z punktu 3

bfaliszek, 2019-04-15:
wstępnie naprawiłem błąd związany z deepsleep. Tutaj jest prereleaseSmogomierz_2.6.6_PMS_build_15.04.2019 https://github.com/hackerspace-silesia/Smogomierz/blob/master/firmware/prereleaseSmogomierz_2.6.6_PMS_build_15.04.2019.bin Jeśli chcesz przetestować czy działa.

skwaruch, 2019-04-16:
Dane PM wysyłają się już poprawnie. Co do tytułu tego tematu to nie podoba mi się ta zmiana szczególnie że tytuł teraz jest całkowicie niepoprawny bo dane się wysyłały.

bfaliszek, 2019-04-16:
@skwaruch .bin prerelease są takimi testowymi betami. Zamknę jak wrzucę .bin do release. Plus Jest problem z MQTT przy deepsleep. Nie odświeża danych. W każdym innym przypadku MQTT i pozostałe serwisy działają poprawnie.

skwaruch, 2019-04-16:
Może by dodać do projektu:
1. pomiar napięcia wraz z wysyłaniem
2. punkt rosy wraz z wysyłaniem
3. wyłączanie PMS sprzętowe (no chyba że działa te programowe? u mnie nie działa)
4. zmiana mocy (TX_POWER)
5. "first scan" - podczas startu uśredniania wartości PM są wpisywane wartości 0 i wartości średnie są nie poprawne, wyeliminowałem to, było to konieczne by usypiać ESP

wparuch, 2019-04-19:
@skwaruch czy bateria 18650 nie daje napięcia 3.7V (vs 5V którego wymaga PMS)? 
Podzieliłbyś się kodem do pomiaru napięcia (1) oraz zmiany mocy (TX_POWER)? Ta zmiana mocy to zmiana mocy WIFI w zależności od jakości sygnału czy coś innego?
Co do (5) - to jest powinno być poprawione w kodzie Smogomierza tutaj: https://github.com/hackerspace-silesia/Smogomierz/commit/bb1a8169f8105d24b84d9aa53bf8e8c320f739ad

skwaruch, 2019-04-19:
18650 li-ion (ja mam powyciągane z laptopa) mają napięcie pracy od 4,1V do około 3,2V, ale już od około 3,5V napięcie zaczyna spadać coraz szybciej. PMS działa dobrze właśnie do około 3,5V poniżej czasami zawyża znacząco pomiar. ESP teoretycznie ma zakres pracy od 3,6V do 3V, ale praktycznie to nawet przy 2,5V działa. Pomiar napięcia realizowany przez ESP podaje wartość o około 0,3V mniejszą niż podaje się na wejście 3V. Przechodząc do sedna, PMS mam podłączony bezpośrednio, a ESP przez diodę prostowniczą (jedynie takie coś znalazłem we swoich szpargałach), która obniża napięcie o około 0,6-0,8V w zależności od poboru prądu przez ESP. Napięcie mierzone przez ESP mam w granicach 3,1V do 2,4V (przy tej wartości usypiam całe urządzenie na 4294967295 (to chyba maksymalną wartość) by zabezpieczyć baterię przed uszkodzeniem).

1. 
ADC_MODE(ADC_VCC); // read supply voltage VCC
float averageVcc = 0;
#define iPM_MAX 10
float Vcc[iPM_MAX];

Vcc[iPM] = (float)ESP.getVcc();

int averageVccf() {
  for (int i=0; i < iPM_MAX; i++){
    averageVcc += Vcc[i];
  }  
  averageVcc = averageVcc/iPM_MAX/1000;
}

4. To zmiana mocy ręczna tak jak to ma miejsce w routerach. Ustawia się to za pomocą funkcji
WiFi.setOutputPower(TX_POWER);
float TX_POWER = 10.00; tą zmienna dałem w defaultConfig.h do ustawienia przez stroną w zakresie 0-20

Jak coś niejasno napisałem to proszę pisać.

Czy działa wam te usypianie programowe PMSa?

bfaliszek, 2019-04-22:
Przed świętami zacząłem testy pracy na 18650. Pomiary(seria 10) co 3 minuty, wysyłanie do Influxa i usypianie. PMS5003 i BME280 i chyba coś jest nie tak. Całość działała tylko około 66 godzin(18.04 o 10:50 start, koniec 21.04 o 4:50)
[image]
[image]
Może w NodeMCU V3 również trzeba mieć rezystor?
https://tinker.yeoman.com.au/2016/05/29/running-nodemcu-on-a-battery-esp8266-low-power-consumption-revisited/

skwaruch, 2019-04-22:
U mnie NodeMCU V3 podczas pracy bierze 70-75mA a jak śpi to 2mA.
Czy wszystkim działa usypianie PMS7003?

wparuch, 2019-04-23:
Dorzucę swoje 3 grosze. PMS pracujący na napięciu innym niż 5V nie daje wiarygodnych wyników (żeby móc je porównywać z innymi urządzeniami).

Co do usypiania - tak na NodeMcu działa właściwie, mam natomiast problem na Wemosie gdzie nie wstaje mi za każdym razem tylko zdarza mu się pomijać cykle albo całkiem zgubić sieć i czekać na ponowną konfigurację.

skwaruch, 2019-04-23:
Tak jak już pisałem do około 3,5V wyniki są takie same jak napięciu 5V.

wparuch, 2019-04-23:
Przy obecnych stężeniach nie zauważysz różnicy.

skwaruch, 2019-04-23:
Nie napisałem że to teraz sprawdzałem.

wparuch, 2019-04-23:
W wielkim skrócie - napięcie niższe niż 5V skutkuje innymi obrotami wiatraczka, co powoduje inny od zaprojektowanego, niewłaściwy przepływ powietrza a więc i niepoprawne pomiary. Jest to podkreślone w dokumentacji oraz zbadane empirycznie. Także jeśli chcesz badać zmiany trendu to ok, ale jeśli chcesz porównywać tak zebrane wyniki z innymi urządzeniami to nie ma szans.

bfaliszek, 2019-04-23:
Producent w specyfikacji zaleca 5V, min. to 4,5V a max, to 5,5V. Trzymałbym się tych przedziałów. 
[PMS7003_datasheet.pdf](https://github.com/hackerspace-silesia/Smogomierz/files/3106412/PMS7003_datasheet.pdf)
[PMS5003_manual_v2-3.pdf](https://github.com/hackerspace-silesia/Smogomierz/files/3106413/PMS5003_manual_v2-3.pdf)
Załadowałem 18650 i podłączyłem na nowo. Ten sam zestaw i takie same ustawienia(seria 10 pomiarów co 3 minuty i wysyłanie do influxdb), jedynie wyłączyłem deepsleep. Zobaczymy jak długo będzie działał.

skwaruch, 2019-04-23:
Też zbadałem empirycznie jakie napięcie może być na PMS.
Jeśli wiatraczek może pracować z różnicą 1V na plus (4,5V do 5,5V) to może też dobrze pracuje z 1V na minus (4,5V na 3,5V)

bfaliszek, 2019-04-23:
Tylko przy 3V masz pewność, że wtedy będzie działać. Dlatego to jest w dokumentacji. Jedno ESP8266 będzie działać przy 2,3V inne przy 2,2V albo jeszcze innej wartości. To co jest w dokumentacji to wartość przy której każde działa poprawnie. W przypadku PMS wiatraczek działa przy 3V co nie znaczy, że pomiary PM mają wtedy jakikolwiek sens.

skwaruch, 2019-04-23:
Mają sens bo to sprawdziłem empirycznie.

bfaliszek, 2019-04-23:
Nie mają i nigdy nie będą mieć. Takie badania robi producent i wie w jakim % przypadków nie mają. Jeśli w Twoim jednostkowym przypadku zadziała, to nie znaczy, że u reszty. Pewność poprawnych pomiarów masz w zakresie 4,5-5,5V.

skwaruch, 2019-04-23:
Ja tylko i wyłącznie piszę o tym co sprawdziłem, czyli o moim, więc proszę mi tu nie wmawiać że pomiar nie ma sensu. W moim np usypianie nie działa.

bfaliszek, 2019-04-24:
To samo 18650 oraz identyczny Smogomierz(NodeMCU V3, BME280 i PMS5003) z tym samym oprogramowaniem. Pomiary(seria 10) co 3 minuty i wysyłanie do InfluxDB. Jedyna różnica w stosunku do poprzedniego pomiaru, to wyłączenie DeepSleep w ustawieniach.
Czas pracy na baterii to około **33 godzin**, czyli 2 razy krócej(Start 23.04 - 9:40, koniec 24.04 - 18:40).

skwaruch, 2019-04-26:
Czy ja mogę zrobić jakaś odnogę od waszego programu?

bfaliszek, 2019-04-26:
Możesz robić, przecież to projekt open source. Zrób sobie fork i możesz zmieniać czy przerabiać dowoli.

bfaliszek, 2019-08-07:
W wersji 2.7.0 chyba wszystko jest OK, więc zamykam issue. Pora na porządki przed jesienią 😉

closed 2019-08-07 by bfaliszek