hackerspace-silesia/Smogomierz #43: Odłączony PMS - nie działa wysyłanie, usypianie ...

skwaruch, 2019-05-03:
Gdy odłączy się PMS urządzenie nie wysyła danych (THINGSPEAK - te sprawdzałem), urządzenie się nie usypia. 
W którym miejscu w kodzie jest taka funkcjonalność zapisana.

bfaliszek, 2019-05-03:
Jaka wersja oprogramowania?
Co rozumiesz przez odłączy?

Wysyłanie do Thingspeak jest tutaj - https://github.com/hackerspace-silesia/Smogomierz/blob/c7866558b5adbc22ec6685b6703ee7123635989b/Smogomierz.ino#L511-L516
sam kod odpowiedzialny za Thingspeak tu - https://github.com/hackerspace-silesia/Smogomierz/blob/master/src/thing_speak.cpp
Jeśli w konfiguracji mamy wybrane brak miernika PM2.5/PM10 to nie wysyła danych danych PM2.5 i PM10. Przy wybranym ale odłączonym może nie wysyłać również(choć nie wiem, nie sprawdzałem), ale to prawidłowe zachowanie. Po co wysyłać dane PM skoro się ich nie mierzy.

skwaruch, 2019-05-03:
Wersja oczywiście aktualna.
Padło zasilanie czujnika (np. wyrwał się kabelek).
W konfiguracji wybrane tak by działało jak jest wszystko podłączone.
Zachowanie nie prawidłowe bo po awarii jednego z czujników zapis innych powinien być i usypianie raczej też.

Z przedstawionego fragmentu kodu raczej nie wynika brak usypiania i wysyłania danych, gdy czujnik ma awarię.

bfaliszek, 2019-05-04:
Po weekendzie to sprawdzę. Nie ma w kodzie sprawdzania poprawności danych z Miernika PM2.5/10. Nie da się za bardzo tego zrobić, bo tutaj albo dostaje się dane z UART albo nie. Gdy ich nie ma to wartości PM wynoszą 0 i nie są zmieniane. Takie powinny być wysyłane do serwisów zewnętrznych przy włączonym w Konfiguracji, ale nie podłączonym mierniku PM. 
Sprawdzanie danych jest tylko dla mierników temp/humi/press tutaj jeśli miernik jest włączony, ale dane są błędne lub ich nie ma(albo gdy jest źle podłączony), to nie są wysyłane.

skwaruch, 2019-05-04:
Nie ma to nie ma, zbędna rzecz, pytanie dlaczego nie działa jak nie ma czujnika.
Pytałem bo nie widziałem jakiegoś sprawdzania czujnika.

bfaliszek, 2019-05-07:
Problem nie leży w wysyłaniu, tylko pobieraniu danych z PMS. W tym miejscu się zatrzymuje przy wypięciu miernika PM(odpinam GND od PMS): https://github.com/hackerspace-silesia/Smogomierz/blob/c7866558b5adbc22ec6685b6703ee7123635989b/Smogomierz.ino#L748-L750
Po ponownym podłączeniu idzie dalej. Dlatego nie wysyła danych, bo "zawiesza" się na pobieraniu danych. Przy czym np. interfejs www w tym czasie działa. Pobiera dane z BME280 itp.

skwaruch, 2019-05-07:
Dlaczego www działa a wysyłanie i usypianie nie?
Jest możliwe że na wersji 2.4.3 się nie wieszało?

bfaliszek, 2019-05-07:
nie, to kwestia biblioteki do obsługi PMS. Od początku jest taka sama. Nie ma w niej możliwości sprawdzenia czy odbierane dane są poprawne. W przypadku wypięcia kabli dane są niepoprawne i odczyt się zawiesza. W zasadzie to nawet nie tyle zawiesza co pauzuje i czeka aż znowu będą poprawne dane. Możesz podłączyć kabel po kilku minutach i wtedy dokończy serię pomiarów. Muszę albo zmienić bibliotekę albo pomyśleć jak to rozwiązać, żeby wykrywać odłączenie przewodów.

skwaruch, 2019-05-19:
Mam mały problem. Jak kompiluje u siebie program baz zmian oczywiście to smogomierz się łączy na 1s i rozłącza i tak w kółko. Jak wgram bin`a ze strony to wszystko jest ok.
Zmieniałam kompa i na nowo ściągałem arduino może czegoś mi brakuje lub coś za bardzo zaktualizowałem lub za mało. Macie może jakiś pomysł?

bfaliszek, 2019-05-20:
Sprawdź z biblioteką płytek esp2866 w wersji 2.5.0. Kilka dni temu na 2.5.1 nie chciało mi się zupełnie skompilować. W wolnej chwili będę to musiał jeszcze raz sprawdzić. Sprawdź też czy masz ustawiony 1MB dla SPIFFS, żeby ESP8266 miało gdzie zapisywać ustawienia.

skwaruch, 2019-05-20:
Dokładnie 5 min temu  na starym kompie zaktualizowałem do 2.5.1 i to samo (nie działa). Chyba coś tam zepsuli.
U mnie się niby kompilowało, ale potem smogomierz nie działa.

bfaliszek, 2019-05-20:
@skwaruch właśnie miałem sprawdzić czy kod się kompiluje i okazało się, że jest już nowa biblioteka płytek 2.5.2 i na niej wszystko działa. Chyba to 2.5.1 miało jakieś błędy. https://github.com/esp8266/Arduino/releases

bfaliszek, 2019-05-20:
Tylko po wgraniu kodu skompilowanego na 2.5.2 Smogomierz wpada w jakiś dziwny loop i ciągle się resetuje. Zmienili trochę sposób zapisu danych w pamięci SPIFFS i to powoduje problem. Na 2.5.0 jest jeszcze ok. Żeby działało na 2.5.2 trzeba będzie dodać parę poprawek w kodzie.

skwaruch, 2019-05-26:
Czym robiliście te schematy elektryczne?

bfaliszek, 2019-05-26:
http://fritzing.org - chyba najbardziej przyjazne schematy dla osób nietechnicznych

skwaruch, 2019-06-04:
Dodałem swoją funkcję do wyliczania napięcia i zastawiałem się pół dnia dlaczego nie działa mi ja jest DEEP_SLEEP. W końcu wpadłem na to że beznadziejnie jest napisana takeSleepPMMeasurements(). Właściwie jak jest załączony DEEP_SLEEP to pętla główna programu nie działa :/