hackerspace-silesia/Smogomierz #44: Niepoprawny build z kodów źródłowych

p-jac, 2019-07-07:
Na początek dziękuję hackerspace-silesia i @bfaliszek - świetny projekt, zmontowałem Wasz smogomierz kilka miesięcy temu i sprawdza się świetnie!

Próbuję go odrobinę wzbogacić (używam smogomierza w mieszkaniu i chcę dołożyć mu czujnik CCS811, LCD 2x16 + pasek 8 LEDów) i mam kłopot z właściwym zbudowaniem pliku .bin.

Działam na wersji 2.7.0 i z gotowym plikiem .bin ściągniętym z gita działa dobrze. Jestem w stanie również skompilować tę wersję u siebie w Arduino i wgrać na płytkę - ale niestety tak przygotowana wersja nie działa (pomimo braku jakichkolweik zmian w kodzie) - sypie się coś przy starcie, bo dioda NodeMCU nie świeci, ewentualnie mruga niemrawo. Taki stan mam zarówno po wgrywce skompilowanego przeze mnie pliku .bin flasherem, jak i bezpośrednio z Arduino.

Po wgraniu oryginalnego pliku .bin z gita, działa dobrze. Binarnie wersje te są różne (.bin z gita ma 497 968 B, a skompilowany u mnie 504 416 B).

Coś mam pewnie nie tak z Arduino, ale nie mogę tego namierzyć. Jedyne co przychodzi mi do głowy to niewłaściwa wersja jakiejś biblioteki - wszystko ściągałem w czerwcu, więc są świeże, ale może właśnie któraś potrzebna jest "nieświeża"? Możecie podrzucić numery wersji bibliotek wykorzystywane przez Was do kompilacji? Ewentualnie inne pomysły, co może być nie tak...

Z góry dzięki za pomoc - jak się uda i moje dodatko zadziałają, dorzucę na gita propozycję kodu do obsługi czujnika CCS811, LCD i LEDów - może komuś jeszcze się przyda :-)

PS Moja konfiguracja: Arduino 1.8.9 na Windows 10, ustawiona płytka ESP-12E, prędkość 115200, CPU 80Mhz, pamięć 4 MB (1 MB SPIFFS), wszystkie debugowania wyłączone. Każdorazowo czyszczę tylko sketch z flasha, żeby nie tracić ustawień. Ustawień Arduino nie ruszałem poza płytką, pamięcią i ściągnięciem bibliotek. Jestem początkujący jeśli chodzi o NodeMCU i szerzej mikrokontrolery, więc może chodzić o jakąś drobnostkę.

bfaliszek, 2019-07-08:
Sprawdź jaką masz wersję biblioteki do obsługi płytek esp8266? Najnowsza to 2.5.2 i na niej się raczej nie skompiluje, bo zmienili coś w obsłudze SPIFFS. Dostępne .bin są skompilowane na 2.5.0 i tam powinno być ok. Jak znajdę trochę wolnego czasu na poprawię kod, żeby działał na 2.5.2.

bfaliszek, 2019-07-08:
Co do CCS811 to w sumie ostatnio trochę z nimi eksperymentuję i w sumie to mogę dodać obsługę, tylko wychodzę raczej z założenie, że Smogomierz to urządzenie, które umieszcza się na zewnątrz, więc pomiary eCO2 i TVOC raczej nie mają sensu. Plus z CCS811 mam problem z ustawieniem baseline, teoretycznie zapisuję po 30 minutach na świeżym powietrzu ustawienia i przywracam je później, a wyniki i tak są zupełnie inne niż z MH-Z19.

p-jac, 2019-07-08:
Dzięki! Na bibliotece płytki 2.5.0 wgrywa się i śmiga bez zarzutu.
Jeśli możesz podrzucić swój kod z CCS811, chętnie potestuję. Mam na razie bardzo wstępną wersję, która tylko wystawia wyniki do bazy influx. Udało mi się to odpalić, ale patrząc na wykres czujnik chyba jeszcze się wygrzewa... rozrzuty pomiarów o setki punktów w odstępie kilku minut.

bfaliszek, 2019-07-10:
CCS811 powinien się nagrzewać jakieś 25 minut i później ma sensowne pomiary. 
Jeśli chodzi o kod, to używałem tej biblioteki https://github.com/maarten-pennings/CCS811

p-jac, 2019-07-11:
No cóż, u mnie po 48-godzinnym burn in CCS811 daje jeszcze bardziej rozedrgane wyniki. Potestuję jeszcze, ale nie wygląda, żeby miał pokazywać coś sensownego (załączam zrzuty gdyby Cię zainteresowały - spadek rano nie daje się powiązać z niczym, czujnik stał 2 metry od uchylonego okna, sądzę że powinien pokazywać w miarę stabilne czyste powietrze...). Używam oryginalnej biblioteki Sparkfun (https://github.com/sparkfun/CCS811_Air_Quality_Breakout), taką mam płytkę.
[image]
[image]

Za to wyświetlacz LCD 2x16 działa dobrze i pasek LEDowy też (pozytywne zaskoczenie, bo zasilam go z linii 3,3V - poniżej specyfikacji). Wszystko (choć nie bez kłopotów z wykorzystaniem pinu D8) udało mi się podpiąć pod NodeMCU. Wrzucę kod jak go trochę posprzątam, może komuś się przyda.

bfaliszek, 2019-08-23:
Hej, w końcu znalazłem chwilę, żeby poprawić kilka rzeczy w kodzie i już mi się kompiluje na esp8266 core 2.5.2, kod jest na razie w branchu dev - https://github.com/hackerspace-silesia/Smogomierz/tree/dev
Wersja 2.7.1 nie wnosi żadnych nowości, po prostu kompiluje się i powinna działać na najnowszych bibliotekach.

bfaliszek, 2019-08-30:
Właśnie udostępniłem poprawiony kod - 2.7.1. Teraz już się kompiluje na najnowszych bibliotekach dla ESP8266 oraz ESP32 ;)

closed 2019-08-30 by bfaliszek