hackerspace-silesia/Smogomierz #45: Wysyłanie danych do Domoticza

Eurynomosek, 2019-07-31:
Można by dodać obsługę Domoticza

firemark, 2019-07-31:
Jasne, możesz dodać ;)
Podeślesz jakieś info o API etc?

bfaliszek, 2019-08-02:
Chciałbym wrócić do rozwoju Smogomierza we wrześniu. Jeśli masz info o tym jak można dodać obsługę Domoticza(jakieś linki, opis ich API itd.), to podrzuć tutaj. Sam nie korzystam, więc nie za bardzo wiem jak to działa.

bfaliszek, 2019-08-05:
Znalazłem taki przykładowy kod do wysyłania danych do Domoticza. Nie wiem tylko czy działa. Postaram się w tym tygodniu zainstalować domoticza na jakimś Raspberry Pi i sprawdzić czy się wysyła. Jeśli będzie ok, to można dodać do Smogomierza
https://diyprojects.io/esp8266-web-client-sending-data-domoticz-tcp-ip-wireless-api-json/#.XUfTLS_UR24

slawek0200, 2020-02-11:
Czekamy :)
to musi działać
http://#IP_SERVER#/json.htm?type=command&param=udevice&idx=X&nvalue=0&svalue=Y
pozdro

jafal99, 2020-02-25:
Cześć,
Dodałem wstępną obsługę wysyłania danych do domoticza przez MQTT. Konfiguracja przez zmienne w defaultConfig, nie ma możliwości konfiguracji przez web bo nie wiem jak się do tego zabrać. Przy okazj zrobiłem mały refaktoring  kodu i wydzieliłem obsługę czujników temperatura/wilgotność/ciśnienie do osobnego pliku.

[0002-Initial-support-for-Domoticz-via-MQTT.patch.zip](https://github.com/hackerspace-silesia/Smogomierz/files/4251157/0002-Initial-support-for-Domoticz-via-MQTT.patch.zip)
[0001-Code-refactoring-generic-functions-for-THP-sensor.patch.zip](https://github.com/hackerspace-silesia/Smogomierz/files/4251158/0001-Code-refactoring-generic-functions-for-THP-sensor.patch.zip)

closed 2021-11-16 by bfaliszek