hackerspace-silesia/Smogomierz #47: Niezabezpieczony AP dostępny

surmacz, 2019-09-22:
Na moim ESP8266 wgrałem wersję 2.7.1 i po podłączeniu do lokalnej sieci, sam chip po pewnym czasie (po którymś podłączeniu do prądu) zaczął wystawiać Access Point o nazwie "ESP_****", niezabezpieczony, po podłączeniu każdy ma możliwość dostępu do danych miernika oraz prawdopodobnie ma dostęp do sieci lokalnej (!) oraz naszego łącza.

Naprawiłem to dodając linię:
```WiFi.mode(WIFI_STA);```
bezpośrednio po każdym wywołaniu ```WiFi.begin```
Po naprawie problem przestał występować, nawet po usunięciu dodanych linii. Nie jestem w stanie zreprodukować więcej błędu.

Tutaj opisany jest podobny przypadek: https://forum.arduino.cc/index.php?topic=501263.0
Myślę, że warto zabezpieczyć ten przypadek, kto wie ile osób nawet nie wie, że udostępnia internet postronnym ;)

bfaliszek, 2019-10-01:
Faktycznie dopisanie `` rozwiązało kilka problemów. Nie tylko z tworzeniem sieci WiFi, ale również z problemami z połączeniem przy słabszym sygnale WiFi.
Wszystkie poprawki w kodzie znajdują się tutaj - https://github.com/hackerspace-silesia/Smogomierz/commit/e3557b4d24135465043debe329b157764ddd83da
Plik .bin z poprawkami jest na razie w branchu dev - https://github.com/hackerspace-silesia/Smogomierz/tree/dev/firmware
Pod koniec tygodnia chciałbym oficjalnie puścić 2.7.2 z tą poprawką.

bfaliszek, 2019-10-04:
https://github.com/hackerspace-silesia/Smogomierz/commit/838b7cd311d618b1357bfc5521e464aa70848a47

closed 2019-10-04 by bfaliszek