hackerspace-silesia/Smogomierz #49: Lat/Lon precyzja float (Airmonitor)

turonl, 2019-10-01:
Jest problem ze współpracą z Airmonitor'em związany z niezgadzającymi się współrzędnymi geograficznymi, które muszą być identyczne z tymi podanymi w ankiecie do AM.
W Smogomierzu przechowywane są jako float - co powoduje, że nie zawsze to co wpiszemy w formularzu zgadza się z tym co jest zapisane/wysyłane.
Teoretycznie z float single precission wpisanie 50.007944 spowoduje zapisanie liczby 50.00794**2**19 co przy konwersji na Stringa da nam prawdopodobnie 50.007942 i taka wartość powinna się pojawić po resecie/ czy tez jest wysyłaną do AM (nie mam teraz wersji Smogomierza z float aby to sprawdzić).

Ponieważ w kodzie Smogomierza Lat/Lon nie są używane w obliczeniach - a tylko konwertowane na String i z powrotem - proponuje zmienić wewnętrzna reprezentacje lat/lon na char[256] lub nieco mniejszy i pozbyć się konwersji do/z float.

Sprawdziłem to na moim Smogomierzu (LATITUDE, LONGTITUFE i ALTITUDE zmienione na char[256]) i współpraca z Airmonitorem działa bezbłędnie. W razie potrzeby mogę przygotować patcha/branch z potrzebnymi zmianami.

Poniżej cytat z korespondencji z twórcami Airmonitor:
"Aktualnie występują pewne nieścisłości w wykorzystywaniu firmware smogomierz, który nieprawidłowo zaokrągla wartości lat/long.
Jeśli korzystasz ze smogomierza to przejdź na wykresy prezentujące pomiary i kliknij prawym przyciskiem myszki na danym wykresie aby skopiować link.
_Przykład: http://metrics.airmonitor.pl:3000/d-solo/000000002/airmonitor?orgId=1&var-latitude=52.294800&var-longitude=20.881800&refresh=1m&panelId=14_
W tym linku zauważysz, że znajduje się lat oraz long, które są wysyłane do serwisu airmonitor. Te wartosci należy wpisać we formularz, a zauważysz samemu, że różnią się one od tych, które sam wpisałeś w smogomierzu."

Pozdrawiam,
Lukasz

bfaliszek, 2019-10-01:
Hej, zmieniłem float na char w kodzie. Jeśli miałbyś chwilę to możesz zerknąć czy masz tak samo - https://github.com/hackerspace-silesia/Smogomierz/commit/a4f62a3b5d9744d983471d0fdd149abdd048d76f 
Plus wrzuciłem .bin z tym kodem do branchu dev - https://github.com/hackerspace-silesia/Smogomierz/tree/dev/firmware

turonl, 2019-10-01:
> zerknąć czy masz tak samo - a4f62a3

Hej,
  Dzięki za szybka odpowiedz - zerknę jak będę w domu.. Dam znać wieczorem
(wstępnie wygląda całkiem podobnie do moich zmian),.
Ja zmieniłem jeszcze altitude na char - tak aby całkiem pozbyć się tych
floatow - ale to nie ma znaczenia dla airmonitor.

Pozdrawiam,
Lukasz

bfaliszek, 2019-10-01:
Zmieniłem altitude z char na int - zajmuje mniej miejsca w pamięci, a tutaj nie jest potrzebna aż taka dokładność. Zresztą już wcześniej przed zapisem w pamięci ESP, zmieniałem tę wartość na inta.
https://github.com/hackerspace-silesia/Smogomierz/commit/0faa4621860a717712ab5d9e7e040202096bdc9d

bfaliszek, 2019-10-04:
https://github.com/hackerspace-silesia/Smogomierz/commit/838b7cd311d618b1357bfc5521e464aa70848a47

closed 2019-10-04 by bfaliszek