hackerspace-silesia/Smogomierz #50: Integracja z aqi.eco

mickelk, 2019-10-05:
znalazłem ciekawą stronę https://aqi.eco która zbiera wykresy, ma swój adres i jest przejrzysta :) da się dodać do smogomierza opcję: wysyłaj do własnego api i opcje do uzupełnienia: adres, ścieżka, port, nazwa użytkownika i hasło żeby móc z tego skorzystać ?

bfaliszek, 2019-10-07:
Przejrzałem projekt i wydaje mi się, że łatwiej, prościej i bardziej sensownie będzie napisać do twórcy(@trekawek), żeby dodał obsługę Smogomierzy przy imporcie danych z madavi.de. Jeśli mamy włączone wysłanie danych do Luftdaten, to trafiają one również do Madavi.de. aqi.eco ma możliwość importu tych danych, tylko jest jeden problem. W kodzie(https://github.com/trekawek/air-quality-info/blob/b7fffc71f8d3138e85fd44f3fa5a65616584bfa8/src/htdocs/model/migration/madavi_migrator.php#L27) początek nazwy ustawiony jest na esp8288-ChipID. Smogomierze mają własną nazwę i są tam zapisywane jako smogomierz-ChipID - Tutaj można podejrzeć pełną listę Smogomierzy wysyłających tam dane pomiarowe do Madavi.de https://www.madavi.de/sensor/graph.php

trekawek, 2019-10-07:
Hej @bfaliszek i @mickelk !

Import z madavi.de jest sposobem aby zainicjalizować bazę danych, ale aqi.eco nie oferuje funkcji okresowego pobierania stamtąd stanu czujników. Preferowałbym rozwiązanie w którym to detektor kontaktuje się bezpośrednio z aqi.eco - w razie czego mogę wystawić API dostosowane pod potrzeby Smogomierza.

Pozdrowienia,
Tomek Rękawek

ookris, 2019-10-07:
A może po prostu dodać w konfiguracji zwyczajne pole wo wprowadzenia własnego URLa który będzie okresowo wykonywany przez Smogomierz?
Coś w stylu
`https://jakas-stron.com/api/zapytanie?czujnik1=[WARTOSC1]&czujnik2=[WARTOSC2]&apikey=[APIKEY]&user=[USER]`
gdzie w miejsce [WARTOSC1], [WARTOSC2] itd. można podstawić dane z czujników podłączonych do smogomierza. Wtedy każdy miałby możliwość skonstruowania swojego własnego zapytania wysyłanego do dowolnego serwera a sam smogomierz po prostu uruchamiałby tylko taki URL.

bfaliszek, 2019-10-07:
Zobaczę jak to będzie działać przy wysyłaniu danych pomiarowych w taki sposób jak jest opisane w "Konfiguracja czujnika" na aqi.eco. Jeśli będzie działać(port 443, więc HTTPS, nie wiem jak sobie z tym poradzi ESP8266).
@trekawek masz może gdzieś przykładowy plik JSON, który można wysłać do aqi.eco?

trekawek, 2019-10-07:
Port 80 i http też jest wspierany. Uwierzytelnienie przy użyciu nagłówka HTTP "Authorization:"

Przykładowy json (używany jest esp8266id i później pola SDS_P1, etc.):
```
{ 
   "esp8266id":"2392534",
   "software_version":"NRZ-2018-123B",
   "sensordatavalues":[ 
      { 
         "value_type":"SDS_P1",
         "value":"6.22"
      },
      { 
         "value_type":"SDS_P2",
         "value":"3.50"
      },
      { 
         "value_type":"BME280_temperature",
         "value":"12.32"
      },
      { 
         "value_type":"BME280_humidity",
         "value":"53.81"
      },
      { 
         "value_type":"BME280_pressure",
         "value":"101526.05"
      },
      { 
         "value_type":"HECA_temperature",
         "value":"20.23"
      },
      { 
         "value_type":"HECA_humidity",
         "value":"30.70"
      },
      { 
         "value_type":"samples",
         "value":"1754298"
      },
      { 
         "value_type":"min_micro",
         "value":"80"
      },
      { 
         "value_type":"max_micro",
         "value":"54954"
      },
      { 
         "value_type":"signal",
         "value":"-69"
      }
   ]
}
```

Lista wspieranych nazw pól:
https://github.com/trekawek/air-quality-info/blob/master/src/htdocs/model/updater.php#L6-L17

trekawek, 2020-01-16:
Pull request z propozycją wsparcia dla aqi.eco: #72.

bfaliszek, 2020-01-16:
Wygląda na to, że wszystko działa poprawnie - https://bfaliszek.aqi.eco

closed 2020-01-16 by bfaliszek