hackerspace-silesia/Smogomierz #51: Dodanie nazwy miejscowości na głównej

swiezaks, 2019-10-07:
Cześć, mały pomysł- można by dodać możliwość wprowadzenia jakichś danych lokalizacyjnych czujnika, np Miejscowość, adres, kto będzie chciał to sobie wprowadzi nazwę i adres kto nie to zostawi puste. 
Wyświetlało by Smogomierz Pogoda i obok miejscowość

bfaliszek, 2019-10-07:
Open Street Maps ma całkiem fajne API do uzyskania takich danych. W Konfiguracji Smogomierza ustawiamy jego współrzędne geograficzne. Można na ich podstawie uzyskiwać sporo danych. Tutaj przykład tego co OSM zwraca do zapytania z domyślnych współrzędnych Smogomierza - https://nominatim.openstreetmap.org/reverse?format=geojson&lat=50.263911&lon=18.995711
Jak widać jest tutaj dokładny adres itp. Można coś z tego wyciągnąć i wyświetlać na głównej. Czy raczej to za dużo i dodać możliwość wpisywania dowolnej nazwy?

swiezaks, 2019-10-07:
Wydaje mi się że samo wprowadzenie nazwy jest wystarczające, w dzisiejszych czasach permanentnej inwigilacji lepiej mniej niż więcej udostępniać:)
Jeszcze jedna funkcjonalność przyszła mi do głowy ściśle powiązana z deepsleep- czy może dodać godzinę i datę ostatniego odczytu? bo wcale to nie jest takie oczywiste.

bfaliszek, 2019-10-07:
ESP8266 nie ma wbudowanego zegara, więc czas trzeba pobierać z zewnętrznego serwera. Przy deepsleep trzeba by to robić przy każdym wybudzeniu się.

closed 2021-11-16 by bfaliszek