hackerspace-silesia/Smogomierz #52: OTA zepsuł AirMonitor - niewidoczna lokacja

Marszczak, 2019-10-07:
Po ostatni OTA moj Smogomierz zniknął z AirMonitora, wiem ze jakies opóźnienie jest dopuszczalne, ale minęły 2 doby i śladu na mapie nie ma.
Czekać cieprliwie?

bfaliszek, 2019-10-07:
Przywróć ustawienia domyślne w Konfiguracji Smogomierza i wpisz jeszcze raz współrzędne. Jeśli to nic nie da, zarejestruj jeszcze raz urządzenie w AirMonitora.

Marszczak, 2019-10-07:
Po przywróceniu domyślnych ze strony pomiary zniknęły pomiary. Zostały tylko wykresy. 

Dziękuje - Marcin

Marszczak, 2019-10-07:
Przed przywróceniem ustawień 

Po resecie 

—
Dziękuje
Marcin

Marszczak, 2019-10-07:
[image]
[image]

bfaliszek, 2019-10-08:
@Marszczak Przywrócenie ustawień domyślnych coś dało, czy nadal nie działa?

kkszysiu, 2019-10-14:
Dzisiaj zgłosiłem do AirMonitora, że wersja 2.7.2 poprawia współrzędne - zmienili oni wiadomość która była wysyłana przy rejestracji (wcześniej przy wysyłaniu Google Formsów wysyłali wiadomość o błędnych współrzędnych).

bfaliszek, 2019-11-06:
Przywrócenie ustawień domyślnych w Konfiguracji i ponowne wpisanie swoich współrzędnych załatwia sprawę.

closed 2019-11-06 by bfaliszek