hackerspace-silesia/Smogomierz #54: Integracja z nettemp

mariano78, 2019-10-08:
Witam, czy istnieje możliwość dopisania w kodzie smogomierza funkcji wysyłających pomiary do Nettempa (www.nettemp.tk)? Można to zrobić za pomocą prostego linku http zawierającego odpowiednio sformatowane pomiary lub za pomoca mqtt pisząc na odpowiedni topic (nie znalazłem ustawień topicu mqtt w smogomierzu).  Czy możemy liczyć na pomoc ?

bfaliszek, 2019-10-08:
@mariano78 masz może jakiegoś linka, albo wiesz dokładnie jak mają byś sformatowane te dane dla MQTT Nettemp? Smogomierz obsługuje MQTT, tylko obecnie nie ma opcji zmiany topic na dowolny. Mogę dodać taką opcję, tylko chciałbym najpierw przetestować czy to zadziała.

mariano78, 2019-10-08:
Już opisuje.  NT musi dostać dane w formacie: 

/ip/nazawaurzadzenia/typ_wartości/jakieś_id

Jeśli posyłam dane z np. ESPEasy to mam w Publishtempate wpisane: 

/%ip%/%sysname%/%valname%/%tskname%

W przypadku smogomierza mogło by to wyglądać następująco: 

192.168.1.1/smogomierz/temp/id_1
192.168.1.1/smogomierz/humid/id_2
192.168.1.1/smogomierz/press/id_3
192.168.1.1/smogomierz/air_pm_10/id_4

NT obsługuje typy danych, które nazywają się:
temp, humid, press, dewpoint, air_pm_25, air_pm_10

bfaliszek, 2019-10-08:
Dzięki, postaram się dodać w kolejnej aktualizacji. Nie będzie z tym dużo roboty, bo obecnie dane formatowane są w takiej formie jak poniżej:

nazwa urządzenia/sensor/PM2.5/wartość
nazwa urządzenia/sensor/PM10/wartość
nazwa urządzenia/sensor/temperature/wartość
nazwa urządzenia/sensor/pressure/wartość
nazwa urządzenia/sensor/humidity/wartość

https://github.com/hackerspace-silesia/Smogomierz/blob/master/Smogomierz.ino#L742-L744

Na koniec mam jeszcze jedno pytanie. NT ma jakiś jeden główny MQTT, czy każdy ma swój własny i ma być możliwość wpisania adresu? Pytam, bo przyznam szczerze, że nie znam zupełnie NT i nie wiem jak działa.

mariano78, 2019-10-09:
Każdy NT ma inny adres IP. Musimy mieć możliwość wpisania w konfiguracji: Adresu IP, portu, usera, hasła. 

Gdyby udało się dodać konfigurację topicu np. na zasadzie takiej jak jest to w ESPEasy (czyli nazwy zmiennych) to powstało by uniwersalne narzędzie mqtt nie tylko do nettempa.

bfaliszek, 2019-10-18:
Myślę, że do miesiąca powinienem się wyrobić. Ostatnio mam mniej czasu na Smogomierz, ale od połowy listopada będę miał na pewno znacznie więcej czasu, chociaż myślę, że zrobię to wcześniej.

swiezaks, 2019-11-22:
Witam:)
Czy jest jakieś światełko w tunelu z poprawką kodu do NT.?
Pozdrawiam

bfaliszek, 2019-11-22:
Jest, tylko teraz zacząłem przepisywać cały webserwer, żeby był asynchroniczny i dopiero jak to skończę, zajmę się dokończeniem nowego designu całego interfejsu użytkownika. Dalej w kolejce jest dodawanie nowych funkcji.

swiezaks, 2020-01-13:
Panie kapitanie czy jest znana data wypuszczenia nowego softu do smogomierza?
:)

bfaliszek, 2020-01-24:
@mariano78 

Czyli, jeśli dobrze rozumiem(jeśli nie, to mnie poprawcie, bo chcę to w końcu dodać do kodu i zależy mi na tym, żeby to MQTT było jak najbardziej uniwersalne), muszę mieć coś takiego:

`MQTT_ADRESS/MQTT_DEVICE_NAME/MQTT_SENSOR_PREFIX/MQTT_VALNAME/MQTT_TSKNAME`

gdzie `MQTT_SENSOR_PREFIX` i `MQTT_TSKNAME` są opcjonalne.

Dla przykładu:
`192.168.1.10/smogomierz-123456/sensor/PM10/id_4`

mariano78, 2020-01-25:
OK, Twoja propozycja jest OK. Jedyne na czym nam zależy, to żeby była możliwość konfiguracji MQTT_TSKNAME. MQTT_DEVICE_NAME - może mieć na sztywno wpisane "smogomierz". Resztę obrobimy u siebie.

bfaliszek, 2020-01-25:
MQTT_DEVICE_NAME - będzie brane z nazwy urządzenia - dla wszystkich pomiarów takie samo.

Muszę dodać okienko z możliwością wpisania: 
- MQTT_SENSOR_PREFIX - dla wszystkich pomiarów taki sam
- MQTT_VALNAME - 7 równych, dla temp, humi, press, pm1, pm2.5, pm10 i airquality. Każde z nich może być(nawet powinno być) inne.
- MQTT_TSKNAME - 7 równych, dla temp, humi, press, pm1, pm2.5, pm10 i airquality. Każde z nich może być(nawet powinno być) inne. 

Spory formularz konfiguracyjny się nam z tego zrobi, bo 15 nowych pozycji do konfiguracji 😄

bfaliszek, 2020-01-25:
OK, to mam pierwszą wersję. Będę bardzo wdzięczny za przetestowanie, bo nie wiem czy tak to ma działać i nie mam za bardzo jak tego przetestować.
Plus na razie wpisywanie tych wszystkich nazw wygląda źle w ustawieniach. Pewnie wizualnie to inaczej rozmieszczę i dodam jakieś testy, ale jak macie chwilę, to sprawdźcie czy to działa.
Plik na ESP8266 z PMSx003
https://github.com/hackerspace-silesia/Smogomierz/blob/dev/firmware/ESP8266/prerelease_ESP8266_Smogomierz_2.7.4_PMS_build_25.01.2020.bin

mariano78, 2020-01-26:
Testujemy i widzę że nie jest dobrze. Czasem po kliknięciu guzika "zapisz" strona się wysypuje. Nie trzyma ustawień czasu wysyłania pomiarów. Mam wpisane w interwale mqqt 1 minutę i czasem po przeładowaniu strony mam wartość 1 a czasem 5. W polach MQTT_VALNAME i MQTT_TASKNAME często pojawia się znak "/" przed wartością i za - tez jest to zjawisko losowe. Mimo wpisania wszystkiego smogomierz i tak posyła dane w formacie  "**Smogly-010B75/sensor/humid/ab/ 17.17**" czyli bez IP. 
[image]

bfaliszek, 2020-01-26:
Czyli dodatkowo posyłać w tym ciągu znaków IP przed nazwą urządzenia?
Jeśli chodzi o te zmiany z 1 na 5 i błędne wyświetlanie, to polecam przywrócić ustawienia domyślne, bo dodałem tutaj masę nowych zmiennych.

bfaliszek, 2020-01-26:
A i jeszcze jedno. Nie musisz samemu nigdzie wpisywać /. Nawet lepiej będzie, gdy tego nie zrobisz.

mariano78, 2020-01-26:
Tak, poprosimy żeby IP posyłało się jako pierwsza zmienna w topicu. Żeby było bardziej uniwersalnie można dodać jakiś checkbox czy posyłać IP czy nie. W NT nam się to przydaje, ale nie każdy inny system tego potrzebuje. Co do znaku slesha "/" to ja go nie wpisywałem, pojawia się ten znak samemu po odświeżeniu strony z ustawieniami.

bfaliszek, 2020-01-26:
To dobrze, bo ogólnie powinno samo rozpoznawać czy są wpisane, czy nie i uzupełniać w zależności od tego czy są potrzebne czy nie. Testowałem kilka możliwości i chyba wyłapałem wszystkie kombinacje. Dlatego nie trzeba wpisywać /. Jeśli jakieś pole zostaje puste, to samo wie, że ma je pominąć i nie dać dwóch / obok siebie.

bfaliszek, 2020-01-26:
Dodałem możliwość dodawania IP do topicu. Do wyboru jest tak albo nie. Testowy build - https://github.com/hackerspace-silesia/Smogomierz/blob/dev/firmware/ESP8266/prerelease_ESP8266_Smogomierz_2.7.4_PMS_build_26.01.2020.bin 
W przyszłym tygodniu chcę jakoś to wizualnie sensownie rozmieścić. Myślałem o zrobieniu takiej tabelki, żeby w każdym wierszu była inna wartość (temp, humi, press, pm1, pm2.5, pm10 i airquality), a w kolumnach dać MQTT_VALNAME i MQTT_TSKNAME.

Co do airquality to raczej nie wiem jak to zrobić w procentach. Ile % jaką ma być wartością. Teraz to po prostu info o tym czy jest zła czy dobra jakość powietrza. Na podstawie europejskich norm dla PM2.5. Stać dane nazwy dla konkretnych wartości PM2.5. Przydaje się to, jeśli ktoś chce dane z MQTT używać do przesyłania informacji do aplikacji Apple Home. Tak poziom zanieczyszczenia powietrza podawany jest słownie, a dopiero po wejściu głębiej są info o liczbowym poziomie zanieczyszczeń.

swiezaks, 2020-01-27:
Właśnie wgrałem na nowo (wgranie espeasy, kasowanie całej pamięci) smogly w ustawieniach samsobie jakieś wartości dodaje, zapisanie ustawień i ustawienia fabryczne, zapomnij sieć można oddalić od siebie bo łatwo jest skasować wifi lub utawienia- ew. dodać monit o potwierdzenie wykonania czynności.
[image]
[image]

bfaliszek, 2020-01-28:
@swiezaks to wersja prerelease, więc jeszcze jej nie testowałem dokładnie i nie jest do końca zoptymalizowana. Wrzuciłem linka, bo zależy mi na tym, żebyście sprawdzili czy to MQTT działa w różnych dziwnych konfiguracjach. Nie korzystam w ogóle z MQTT, więc nie mam za bardzo jak sprawdzić samemu. 
Cały ten fragment strony z możliwością wpisywania ustawień MQTT będzie finalnie wyglądał zupełnie inaczej, tylko nie wiem czy to już wszystkie potrzebne parametry czy mam tego dodać jeszcze więcej.

mariano78, 2020-01-28:
Nie jesteśmy w stanie sprawdzić czy działa bo albo się nie da zapisać tych ustawień - klikam zapisz i strona się wysypuje, albo pojawiają się dziwne "krzaki" w polach. Temat jest mocno niestabilny i mocno losowy. Nie jestem w stanie uzyskać stabilnego punktu w którym smogomierz zapisze poprawnie te ustawienia.

bfaliszek, 2020-01-30:
@mariano78 ok, dzięki za konkretne info. W weekend nad tym siądę i dam całą konfigurację MQTT na oddzielną podstronę w konfiguracji serwisów. Wtedy powinno być lepiej.

swiezaks, 2020-01-30:
jeżeli uważasz, że będzie Ci łatwiej testować to mogę Ci podesłać dane do zalogowania się na smogly oraz moje rasperry pi z NT. i tak za parę dni będę stawiał nowy system na pi4 to nawet jak mi coś podrzucisz do systemu to i tak będzie skasowane (oczywiście żartuje:)  )

bfaliszek, 2020-02-01:
Sprawdźcie jak ta wersja. Przerzuciłem całą konfigurację MQTT na oddzielną stronę, plus zwiększyłem ilość pamięci na plik konfiguracyjny o 1/2 - https://github.com/hackerspace-silesia/Smogomierz/blob/dev/firmware/ESP8266/prerelease_ESP8266_Smogomierz_2.7.4_PMS_build_1.02.2020.bin

Eurynomosek, 2020-02-01:
Update error: ERROR[10]: Magic byte is wrong, not 0xE9

swiezaks, 2020-02-01:
oki, nowa wersja wgrana, czeka na mariano78 aż znajdzie chwilę czasu i popatrzy

bfaliszek, 2020-02-01:
> Update error: ERROR[10]: Magic byte is wrong, not 0xE9

Wyłącz z prądu i włącz ponownie urządzenie. Możesz też spróbować przenieść urządzenie bliżej routera. Wystąpił u Ciebie jakiś błąd w czasie przesyłania pliku .bin do urządzenia.

mariano78, 2020-02-02:
Jest lepiej ale nie idealnie :)
przy posyłaniu airquality nie dodaje się przedrostek "sensors"
nie posyła się wartość ciśnienia, ale to chub będzie spowodowane brakiem czyjnika ciśnienia atmosferycznego
Na początku topicu przed adresem IP poprosimy żeby był znak "/"

W dalszym ciągu czasem w formularzu wstawiają się znaki "/" na początku lub na końcu. 
[image]
[image]

bfaliszek, 2020-02-02:
Jeśli nie masz pomiarów ciśnienia, to dane nie są wysyłane. Ogólnie to jakie dane możesz wysyłać zależy od tego jakie masz czujniki, więc to jest ok.
Dodałem "/" na początku i "sensora" do airquality. 
https://github.com/hackerspace-silesia/Smogomierz/blob/dev/firmware/ESP8266/prerelease_ESP8266_Smogomierz_2.7.4_PMS_build_2.02.2020.bin

mariano78, 2020-02-02:
I teraz jest OK, pozostaje tylko kwestia tych przypadkowych znaków "/", ale to w niczym nie przeszkadza w komunikacji. 

Mam jeszcze jedno pytanie - wartość airquality jest z czujnika wysyłana jako zmienna tekstowa, czy jest to liczba gdzieś przeliczana i zamieniana na text ?

bfaliszek, 2020-02-02:
@mariano78 wielkie dzięki za pomocy i cały feedback związany z MQTT. Teraz tylko wizualnie lepiej to ogarnę w panelu i będzie ok. 

Jeśli chodzi o "/" to chyba wiem dlaczego się pojawiają. Oprogramowanie samo rozpoznaje czy w danym polu jest jakaś wartość czy nie. Jeśli nie ma, to inaczej rozstawia "/" w całym topicu, tak żeby nie było np. /// pod rząd. Jeśli wpiszemy jakąś wartość, to samo oda na początku, lub na końcu /, tak żeby topic miał sens. Stąd pewnie te / w niektórych polach. 

Co do wartości tekstowych, to są ustalane na bazie europejskich norm jakości powietrza - https://www.eea.europa.eu/themes/air/air-quality-index
W kodzie to jest tutaj: https://github.com/hackerspace-silesia/Smogomierz/blob/master/Smogomierz.ino#L1030-L1042

mariano78, 2020-02-03:
Nie ma za co dziękować, jak tylko mam czas to chętnie pomogę. 

A czy dałbyś radę zrobić jeszcze jeden checkbox odnośnie averagePM25,  tak żeby można było wybrać czy posyła tekstowo czy posyła właśnie wartość averagePM25 w postaci liczbowej ?

bfaliszek, 2020-02-03:
@mariano78 przecież jest też PM2.5 w wersji liczbowej, masz to nawet na screenie, który wcześniej wstawiłeś. To air_pm_25 to właśnie averagePM25 w postaci liczbowej.
> [image]
Przemyślałem trochę jak to najwygodniej umieścić w panelu użytkownika i chyba zmieniłbym jeszcze jedną rzecz. Zamiast wpisywać oddzielnie każdy z elementów topiku, dam po prostu większe okienko, gdzie będzie można samemu wpisać co się chce. Wtedy będzie można mieć zupełnie inny topic dla każdej wartości.
Będzie do wyboru: IP/brakIP, wybór czy chcemy nazwę urządzenia czy nie i później duże okienko, gdzie wpisuje się wszystko co się chce(np. sensor/humid/ab/) i na koniec będzie wyświetlana aktualna wartość. Zrobiłbym coś takiego dla każdej wartości(Temperatura, Wilgotność, Ciśnienie itd.)

mariano78, 2020-02-03:
Heh byłem przed kawą :)
Twoja propozycja wydaje się być ok.

bfaliszek, 2020-02-03:
Coś takiego:
[image]
W zależności od tego czy mamy wybrane dodawania adresy IP i nazwy urządzenia do topicu, tak zmienia się treść wpisów. Plus dane na końcu są aktualizowane przy lądowaniu strony

mariano78, 2020-02-03:
Gitara :)

bfaliszek, 2020-02-03:
Dodałem te zmiany. Możecie sprawdzić dla pewności czy wszystko działa poprawnie. Jeśli tak, to nie będę tego na razie zmieniał. https://github.com/hackerspace-silesia/Smogomierz/blob/dev/firmware/ESP8266/prerelease_ESP8266_Smogomierz_2.7.4_PMS_build_3.02.2020.bin
Zostało mi jeszcze poprawić obsługę InfluxDB i nowa wersja oprogramowania będzie gotowa.

mariano78, 2020-02-04:
Sprawdziłem, wygląda, że jest OK. Dziękujemy.

bfaliszek, 2020-02-06:
https://github.com/hackerspace-silesia/Smogomierz/releases/tag/v2.7.5

closed 2020-02-06 by bfaliszek