hackerspace-silesia/Smogomierz #56: Pomiar napięcia zasilania Smogomierza

porebskitomasz, 2019-10-15:
Witam.
Zmajstrowałem Smgomierza zgodnie z opisem. 
Wszystko działa wyśmienicie. 
Ja swoją wersję zasilam niezależnie z ogniw 3x18650 o łącznej pojemności około 8000mAh.  Jednocześnie wykorzystałem do doładowania ich dwa niewielkie panele fotowolaticzne o napięciu do 6V- niestety nie potrafię określić ich mocy.
Moje pytanie/propozycja brzmi: Czy byłaby możliwość dopisania funkcji pomiaru napięcia akumulatorów, i wysyłania na przykład do serwisów np http://thingspeak.com, lub żeby było widoczne na głównej stronie z pomiarami Smogomierza? 
Tomek

turonl, 2019-10-22:
Hej,
  Czy masz tam jakiś stabilizator czy po prostu podłączasz te ogniwa do 5v ESP8266?
Z odpowiednim dzielnikiem powinno się dać użyć wejścia ADC Esp8266 - trochę rozbieżne są tylko opisy co do zakresu napiec tego ADC i obecności tegoż dzielnika na płytce ESP.
Tu przykład : https://www.engineersgarage.com/esp8266/nodemcu-battery-voltage-monitor/
Pytanie czy jest potrzeba dodać do Smogomierza coś takiego na stale (jako opcja)...

Riventros, 2020-01-25:
Hej, 
Na jak długo starcza zasilania? Ogniwa zmieściły się w zalecanej obudowie? Czy twój układ pracuje na zewnątrz?

Pozdrawiam.

m-dudzinski, 2020-02-17:
Cześć, ja też jestem już po budowie własnego egzemplarza, wszystko śmiga aż miło :) Zamówiłem też sobie akumulatorki 18650 razem z modułem ładowania ([link](http://pl.aliexpress.com/item/32966489898.html?spm=a2g0o.productlist.0.0.619b3bd0bmf58O&algo_pvid=83c4a9f4-1ede-41bd-b2e2-0349f699f4e7&algo_expid=83c4a9f4-1ede-41bd-b2e2-0349f699f4e7-3&btsid=0b0a0ad815819435168255494e8852&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)), żeby móc zamontować urządzenie na zewnątrz bez konieczności podciągania do niego kabla. 

Czy jest możliwość aby te dane były wysyłane tak jak pomiary np PM2.5 do serwisów (ja używam thingspeak)? Jak podejść do tego tematu?

closed 2021-11-16 by bfaliszek