hackerspace-silesia/Smogomierz #57: Problemy z czujnikiem PM7003

kkszysiu, 2019-10-17:
Witam.

Wydaje mi się, że mam jakieś problemy z moim czujnikiem albo integracją z nim.
Otóż raz na jakieś 24 godziny czujnik zaczyna wysyłać wartości -1.

[image]
[image]

Występowanie wygląda zupełnie losowo. Mogę spokojnie podpiąć się do interfejsu WWW ale i tam widzę jedynie wartości -1.
Jedyne co wtedy pomaga to reset smogomierza.

Spotkał się ktoś wcześniej z takim problemem?

bfaliszek, 2019-10-17:
Sprawdź czy kabelki są mocno dociśnięte. Wydaje mi się, że to może być problem z podłączeniem.

kkszysiu, 2019-10-21:
@bfaliszek, z kabelkami wszystko wydaje się być ok. Dodałem restart esp co godzinę i wygląda to teraz czasami tak:
[image]

Więc nadal od czasu do czasu problem występuje ale restart pomaga.

Ciekawe czy to może być jakoś związane z tym, że używam ESP32. Testowałeś może Smogomierz na ESP32?

bfaliszek, 2019-10-21:
Mam jeden na ESP32, ale nie używam go na co dzień, więc ciężko mi powiedzieć jak to wygląda na dłuższą metę. Jutro go gdzieś podepnę i zobaczę jak będzie działał.

kkszysiu, 2019-10-22:
Dzięki @bfaliszek. Póki co dodałem restart jeśli pomiary wszystkich PM miały wynik -1.
W takim przypadku dumpuje też wszystkie pomiary etc. i wysyłam sobie to do mojego lokalnego serwera. 

Może uda mi się ustalić co jest nie tak. Przez ostatnie 24 godziny jednak problem nie wystąpił.

bfaliszek, 2019-10-22:
a tak z ciekawości, jak masz ustawioną częstotliwość wykonywania pomiarów? Oraz ile robisz pomiarów do uśrednień?

kkszysiu, 2019-10-22:
@bfaliszek, tutaj:
[image]

closed 2019-11-20 by kkszysiu