hackerspace-silesia/Smogomierz #58: Pomiary PMS7003 vs SDS011

skwaruch, 2019-10-19:
Do przemyślenia :)
SDS011:
[image]
PMS7003:
[image]

bfaliszek, 2019-10-19:
PMS7003:
"±10% @ 100 ~ 500 μg/m³
±10μ g/m³ @ 0 ~ 100 μg/m³"
[PMS7003_manual.pdf](https://github.com/hackerspace-silesia/Smogomierz/files/3746439/PMS7003_manual.pdf)

SDS011:
"Maximum of ± 15% and ±10μg/m3"
[SDS011_laser_PM2.5_sensor_specification-V1.3.pdf](https://github.com/hackerspace-silesia/Smogomierz/files/3746440/SDS011_laser_PM2.5_sensor_specification-V1.3.pdf)

hali71, 2020-02-14:
Mam pytanie o kalibracje czujnika. Jakie jest źródło obliczania współczynnika calib:
```
      if (int(BMESensor.temperature) < 5 or int(BMESensor.humidity) > 60) {
        calib1 = float((200 - (BMESensor.humidity)) / 150);
        calib2 = calib1 / 2;
        calib = calib2;
      } else {
        calib = calib1;
```
Szukałem jakiegos potwierdzenia tej kalibracji w necie ale niestety nie znalazłem. A od jakiegoś czasu zastanawiam się jak korygować pomiary np podczas duzej wilgotności lub mgle. Uzywam czujnika PMS7003.

bfaliszek, 2020-02-14:
calib obliczany jest na podstawie pomiarów wilgotności powietrza. Później pomiary z PMS są mnożone razy ten współczynnik. Wypracowałem go na podstawie pomiarów z kilku różnych urządzeń pomiarowych, w tym profesjonalnych pożyczonych od alarmu smogowego.

black-red-dot, 2021-02-11:
Czy wspołczynnik kalibracji obliczany jest w ten sposób:
`float((200 - (BMESensor.humidity)) / 150);` calib1
czy jeszcze dodatkowo dzielony jest przez 2:
`float((200 - (BMESensor.humidity)) / 150)/2;` calib2

z kodu, od miejsca:
`// Automatic calibration - END`
`if (!strcmp(THP_MODEL, "BME280")) {
    calib = calib1;`
wynika, że dodatkowe dzielenie jest pomijane

bfaliszek, 2021-02-11:
Te współczynniki obliczane są zawsze, ale w zależności od ustawień, są używane albo nie. Choć nie do końca wiem, o który fragment kodu Ci chodzi. Zamiast tak cytować kawałek linijki, możesz wrzucić linka do niej w konkretnym pliku w repozytorium. Wtedy będę bardziej wiedział o które miejsce chodzi.

black-red-dot, 2021-02-11:
Chodzi mi to, że najpierw współczynnik calib przyjmuje wartość calib2 [https://github.com/hackerspace-silesia/Smogomierz/blob/master/Smogomierz.ino#L1890-L1892](url) ale za chwilę zawsze zmieniany jest na calib1 [https://github.com/hackerspace-silesia/Smogomierz/blob/master/Smogomierz.ino#L1926-L1927](url)

closed 2021-11-16 by bfaliszek