hackerspace-silesia/Smogomierz #59: Zawieszanie SDS011

skwaruch, 2019-11-07:
Coś nie ma szczęścia do czujników PMS7003 nie wyłącza się softwear'owo a SDS011 wiesza mi się średnio co dzień (smogomierz pokasuje ciągle tą samą wartość). Ponowne uruchomienie czujnika pomaga.

bfaliszek, 2019-11-07:
Oprogramowanie 2.7.2 i ESP8266?

skwaruch, 2019-11-07:
Tak

piotrrekkk, 2020-02-07:
Jakaś aktualizacja odnośnie tego buga? Też się z nim spotkałem.

Kriztower, 2020-02-07:
[image]
U mnie podobnie. Wiesza się minimum codziennie, jak nie kilka razy dziennie. Po restarcie kilka godzin spokoju i znowu.

slawek0200, 2020-02-07:
Ja również potwierdzam, restart co kilka godzin. SDS011 się zawiesza.

theundefined, 2020-02-08:
U mnie na wemos d1 + sds11 +dht22 działa stabilnie. 2.7.2.
Jedno co to mam wrażenie że zaniża wielkości pomiarów - trend jest zgodny z tym co okoliczne czujniki syngeos i looko2, ale wielkosci niższe.

slawek0200, 2020-02-09:
U mnie płytka CH340 NodeMcu V3, +SDS011 +BME280 są problemy z zawieszaniem. Wymienię na  płytkę WEMOS D1 i sprawdzę.

bfaliszek, 2020-02-09:
> U mnie na wemos d1 + sds11 +dht22 działa stabilnie. 2.7.2.
> Jedno co to mam wrażenie że zaniża wielkości pomiarów - trend jest zgodny z tym co okoliczne czujniki syngeos i looko2, ale wielkosci niższe.

Ogólnie looko2 używa mierników PMS5003. Najbardziej zauważalna różnica pomiędzy PMS a SDS to różnica w wykrywaniu poziomu zanieczyszczeń przy ich niskim poziomie. SDS lepiej wykrywa poziomy do około 15ug/m3. PMSy zazwyczaj pokazują 0-2, 5 albo od razu 10ug/m3. SDS jest po prostu bardziej czuły przy niskich stężeniach pyłu. To może być powodem rozbieżności. Plus dht22 często podaje wilgotność z czapy, więc może też wpływać na pomiary PM, jeśli masz włączoną kalibrację w ustawieniach.

bfaliszek, 2020-02-09:
> U mnie płytka CH340 NodeMcu V3, +SDS011 +BME280 są problemy z zawieszaniem. Wymienię na płytkę WEMOS D1 i sprawdzę.

Wydaje mi się, że zmiana z NodeMcu na Wemosa nie powinna mieć żadnego wpływu. Sprawdzę w wolnej chwili jak całość działa z SDSem. Możliwe, że jest jakiś błąd w bibliotece do obsługi SDSów, bo ten błąd nie występuje w innych wersjach oprogramowania, a one różnią się tylko obsługiwanymi miernikami pyłu.

Kriztower, 2020-02-09:
> Ogólnie looko2 używa mierników PMS5003. Najbardziej zauważalna różnica pomiędzy PMS a SDS to różnica w wykrywaniu poziomu zanieczyszczeń przy ich niskim poziomie. SDS lepiej wykrywa poziomy do około [...]

Jeszcze się u mnie na zewnątrz nie zdąży o żeby czujnik (sds011) pokazał <10 😀

Kriztower, 2020-02-10:
Niestety znowu to samo😭
Widać dwa zawieszenia, a po restarcie znowu mierzy przez kilka godzin i znowu zawias. 
A ciekawe że temperaturę i wilgotność pokazuje dobrze cały czas. 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

skwaruch, 2020-02-10:
Nic w tym ciekawego że jest temp. bo zawiesza się SDS011, który temp. nie mierzy.

theundefined, 2020-02-10:
Może inny zasilacz?

Kriztower, 2020-02-10:
> Może inny zasilacz?

Chodzi bezpośrednio z płytki. 

> Nic w tym ciekawego że jest temp. bo zawiesza się SDS011, który temp. nie mierzy.

No tak tylko dlaczego? Wina czujnika? Przecież po restarcie działa ok.

theundefined, 2020-02-10:
Może zwiesza się SDS przy problemach z zasilaniem? Strzelam. Jak nie masz jak zmienić SDS spróbuj jakiś mocniejszy zasilacz użyć :-)

bfaliszek, 2020-02-11:
@Kriztower tutaj opcje są dwie, a w zasadzie to trzy. Dane o temperaturze są wysyłane, więc urządzenia działa. Pomiary temperaturze są dobrze wykonywane, więc jedyne co nie działa to wykonywanie pomiarów przez SDS. Stąd trzy opcje:

+ biblioteka do obsługi SDS ma jakieś błędy i co jakiś czas nie działa

+ trzeba wymienić kabelki do SDS, bo nie można się z nim czasami połączyć

+ jest problem z zasilaniem, lub sam SDS jest wadliwy

skwaruch, 2020-02-12:
Zmieniłem wczoraj zasilacz na mocniejszy i o ciut większym na napięciu (5,12V wcześniej 4,9..V).
Też się zawiesiło :( .

Kriztower, 2020-02-21:
U mnie cały czas się zawiesza. Raz chodzi pół dnia raz półtora dnia, ale zawsze się zawiesi. Czy dałoby radę dopisać linijkę żeby np. co 10 min nastąpił restart samego czujnika, poprzez odłączenie do niego zasilania? 

[image]

MiG-41, 2020-12-19:
U mnie jeszcze inna reakcja , na wsadzie ESP8266_Smogomierz_2.7.2_SDS011_build_4.10.2019.bin działał 5 dni i było ok , zaktualizowałem automatycznie do najnowszej 2.7.9 SDS i wtedy zaczęły się zawieszania czytnika. Zwykły restart nie wystarczał , trzeba było odłączać na chwilę zasilanie.
Teraz wgrałem ESP8266_Smogomierz_2.7.3_SDS011_build_17.01.2020.bin , na razie od 12godzin jest ok ( choć tu mam wielkie problemy z ustawieniem konfiga , uruchamiam jedno , i przestawia się do defaultowych ustawien , czasami cos przestawie , a np jezyk sie cofa na angielski..)

closed 2021-11-16 by bfaliszek