hackerspace-silesia/Smogomierz #60: Projekt nie kompiluje się z ESP8266 2.6.0

surmacz, 2019-11-09:
Po aktualizacji paczki "esp8266" by ESP8266 Community do wersji 2.6.0 pojawia się błąd:
```no matching function for call to 'SoftwareSerial::SoftwareSerial(int, int)'```

bfaliszek, 2019-11-09:
Użyj gotowych plików .bin albo esp8266 w wersji 2.5.2. Przy okazji kolejnej aktualizacji oprogramowania dla Smogomierze, dostosuję kod do wersji 2.6.0

bfaliszek, 2019-11-09:
W branchu /dev dodałem już poprawki dla 2.6.0. Okazuje się, że wystarczy zmienić 2 linijki w pliku Smogomierz.ino 
https://github.com/hackerspace-silesia/Smogomierz/commit/310d6c33063fa306b3dbe58897e28cb8584c1a21#diff-953370ae4a9256d33d95c5429b45f5fcL135

bfaliszek, 2019-11-09:
https://github.com/hackerspace-silesia/Smogomierz/commit/4305d1e263a50742349223baecb486d74bc18557

closed 2019-11-09 by bfaliszek