hackerspace-silesia/Smogomierz #61: Po podłączeniu do zasilania PMS7003 zasuwa bez przerwy do czasu kolejnego interwału

surmacz, 2019-11-12:
Jeśli mam pomiary robione co 15 minut, to przez pierwsze 15 minut po podłączeniu do prądu nie ma pomiaru, ale czujnik cały czas chodzi. Tak musi być?

bfaliszek, 2019-11-12:
A jak masz ustawioną opcję "Ciągłe pomiary PM"? Jeśli jest włączona, to miernik chodzi ciągle, jeśli jest wyłączona, to jest usypiany pomiędzy seriami pomiarów.

surmacz, 2019-11-12:
Ciągle pomiary: NIE
Po pierwszych 15 min. działa OK i się wyłącza, tylko te pierwsze 15 minut ciągle mieli.

surmacz, 2020-01-25:
To nie przypadkiem tutaj jest ustawione?:
https://github.com/hackerspace-silesia/Smogomierz/blob/v2.7.3/Smogomierz.ino#L423
`pms.sleep()` jest dopiero robione po pierwszym pomiarze.

bfaliszek, 2020-01-25:
`if (FREQUENTMEASUREMENT == true) {
      pms.wakeUp();
      delay(500);
      pms.activeMode();
    } else {
      pms.passiveMode();
      delay(500);
      pms.sleep();
    }`

ten `pms.wakeUp();` jest robiony tylko jeśli masz włączone ciągłe pomiary(FREQUENTMEASUREMENT), jeśli są wyłączone, to wykonywany jest kod poniżej linijki `    } else {`

closed 2021-11-16 by bfaliszek