hackerspace-silesia/Smogomierz #63: InfluxDB

murowy, 2019-11-26:
Bardzo proszę o pomoc w skonfigurowaniu InfluxDB.
Jak ThingSpeak kumam, tak  InfluxDB z jego Buckets/Telegraf nie mogę rozgryźć. 
Próbuję pobawić się docelowo Grafaną.
Może dałoby się stworzyć instrukcję tak jak do ThingSpeak, chociaż skrótowo.
Z góry dziękuję.

ookris, 2019-11-29:
Ale czego nie kumasz w Influx? Jak stworzyć bazę danych? Jak zapisywać do nie dane ze smogomierza? 

W zasadzie do działania potrzeba tylko samego silnika InflixDB bez nakładek tylu Telegraf czy innych. 
Proces wygląda tak:
- instalujesz InfluxDB
- tworzysz bazę danych 
- tworzysz sobie user (oczywiście z hasłem) który nie jest administratorem i dajesz mu zezwolenia na READ/WRITE
- konfigurujesz smogomierz tak aby wysyłał dane do bazy influx 
Tu masz w zasadzie wszystko na temat autoryzacji, tworzenia baz danych, użytkowników oraz przydzielania praw dostępu https://docs.influxdata.com/influxdb/v1.7/administration/authentication_and_authorization/

Jak chcesz mieć ładne wykresy to oczywiście możesz zainstalować sobie Chronograf lub ewentualnie Grafanę. Ot cała "konfiguracja"

murowy, 2019-11-30:
Czyli do InfluxDB Cloud się nie podłączę, tak? Rozumiem, że InfluxDB musiałbym sobie zainstalować na swoim prywatnym kompie/serwerze. Myślałem, że to jest tak jak z ThingSpeak, wszystko online bez konieczności posiadania własnego serwera.
Może dało by się jednak rozwinąć funkcjonalność Smogomierza o przesłanie danych na InfluxDB Cloud?

ookris, 2019-11-30:
Nie wiem jak działa InfluxDB Cloud ale wydaje mi się, że nie powinno być problemów z połączeniem. Podajesz adres serwera, nazwę bazy danych, nazwę użytkownika, hasło i w zasadzie wszystko powinno działać. 
Zawsze możesz sobie postawić Influxa na AWS gdzie przez rok masz jedną instancje za darmo ewentualnie za 5 USD/m-c na Digital Ocean.

michland, 2021-02-16:
Co do InfluxDB mam taki problem, że Smogomierz przesyła mi 1 log do bazy i to jest na tyle, wtedy frontend jest niedostępny, wszytko pada, dane do innych serwisów np. aqi, thingspeak się nie przesyłają. Ping do ESP jest normalnie.  Jedyna opcja odłączenie  z prądu i zmiana na NIE wysyłaj do InfluxDB :(

closed 2021-11-16 by bfaliszek