hackerspace-silesia/Smogomierz #64: ThingSpeak gdy kanał nie jest publiczny

4zriel, 2019-12-05:
Hej,
Od wczoraj szczęśliwy posiadacz smogomierza - wielkie dzięki za wszystko! To tak tytułem wstępu :) 

Zauważyłem, że gdy kanał z thingspeak jest ustawiony na private to nie zadziałają wykresy z domyślnej stronki smogomierza, albo inaczej, nie zadziałają o ile nie mamy w cookies tokena z thingspeak (incognito czy inna przeglądarka). Dawno nie kodziłem w cpp więc rzuciłem na szybko na to okiem i imho brakuje read apiKey w adresie w pliczku html-content.h (no i w configu pewnie) -> mogę się mylić co do miejsca. 

np. https://thingspeak.com/channels/XXXXXX/charts/5?bgcolor=%23ffffff&color=%23d62020&dynamic=true&results=60&title=Ci%C5%9Bnienie&type=line&update=15&api_key=API_KEY

bfaliszek, 2019-12-05:
Gdy kanał jest prywatny, to obecnie wykresy nie są wyświetlane. Tak jak napisałeś. Pewnie można to zrobić w taki sposób jak napisałeś. Poprzez dodanie `read apiKey` oraz możliwości wpisania go w Konfiguracji Smogomierza. Postaram się to dodać w przyszłości.

4zriel, 2019-12-05:
@bfaliszek a pull request są ok? :) bo może w wolnej chwili (przy 2jce dzieciaków jest z tym śmiesznie) dodam i wrzucę ;)

bfaliszek, 2019-12-06:
@4zriel zawsze możesz zrobić PR. Na pewno to przyspieszy sprawę. Ogólnie to mam kilka rzeczy do poprawienia w obecnym kodzie, bo strona z Konfiguracją się nie ładuje. W ostaniej wersji "ESP8266 core for Arduino" zmienił się trochę sposób działania WebServera. Kod się kompiluje, ale strona z Konfiguracją nie działa. Po skompilowaniu na starszej wersji "ESP8266 core for Arduino" wszystko działa poprawnie. Dlatego pliki .bin ze skompilowanym oprogramowaniem działają poprawnie.

bfaliszek, 2020-01-23:
https://github.com/hackerspace-silesia/Smogomierz/commit/f3c2a308787b0e4410c367101dae0a911b15894c
Bin dla ESP8266 z PMSx003 - https://github.com/hackerspace-silesia/Smogomierz/blob/dev/firmware/ESP8266/prerelease_ESP8266_Smogomierz_2.7.4_PMS_build_23.01.2020.bin
Oficjalnie wsparcie dla nie publicznych kanałów thingspeak pojawi się w 2.7.4. Jeśli ktoś chce przetestować, to można już instalować wersję prerelease.

bfaliszek, 2020-02-06:
https://github.com/hackerspace-silesia/Smogomierz/releases/tag/v2.7.5

closed 2020-02-06 by bfaliszek