hackerspace-silesia/Smogomierz #68: Prośba o pomoc  przeniesieniem i2c.

majstercar, 2019-12-31:
Proszę o pomoc zmiany kodu do obsługi czujnika BME280 z pinów GPIO 0 na 14 SDA oraz z 02 na 12 dla SCL - Jestem zielony w C++ i pewnych rzeczy nie rozumiem chociaż sam program jest bardzo dobrze skomentowany.
Pozdrawiam 
Sebastian

bfaliszek, 2020-01-02:
Hej!
Zakładam, że masz ESP8266.
W tej linijce https://github.com/hackerspace-silesia/Smogomierz/blob/master/Smogomierz.ino#L369
uruchamiamy obsługę BME280 na pinach D3, D4. D3 to GPIO-0, a D4 to GPIO-2. Stąd liczby w nawiasach. Możesz zmienić je na inne. Wtedy powinna zadziałać obsługa BME280 na innych pinach. 
[image]

majstercar, 2020-01-02:
Okazało się że mój to BMP - robiłem ta że  wstawiałem w liniach 369, 375 różne wartości a na koniec Wire.begin() - oczywiście PMS przemapowałem z oryginału na (0, 2) i rusza bez problemu. Program ze strony https://myesp8266.blogspot.com/2016/12/bmp280-and-esp8266.html bez problemu w monitorze pokazuje temp i ciśnienie a Smogomierz nie chce. Jeszcze jedno przy kompilacji zawsze pokazuje mi ostrzeżenia z biblioteki wire.h ale przechodzi. Nie mam już pomysłu co dalej robić (adres i2c to 0x76) Pozdrawiam i dziękuję za odpowiedź.
Sebastian

bfaliszek, 2020-01-03:
@majstercar sprawdź czy działa https://github.com/adafruit/Adafruit_BMP280_Library - ta bibliotek jest użyta w Smogomierzu. Możliwe też, że jakaś inna biblioteka jest na tych samych pinach i dlatego nie działa.

majstercar, 2020-01-03:
BINGO - wystarczyło pozamieniać w bibliotece dołączonej do smogomierza (Adafruit_BMP280.h) adresy i2c miejscami jako podstawowy umieściłem 0x76, a alternatywny 0x77. Wszystko działa

closed 2020-01-03 by majstercar