hackerspace-silesia/Smogomierz #69: Dodatkowe czujniki

jakubcichocki, 2020-01-12:
Witam,

potrzebuję dołożyć dodatkowe czujniki do Smogomierza - na początek dołożyłem zwykły fotorezystor aby mierzyć ilość światła naturalnego na dworze. 

Pod A0 podłączyłem fotorezystor - odczyt wypada zgodnie z moimi oczekiwaniami, przekształcenie wartości i wyrzucenie ich do konsoli również . Niestety wrzucenie 

`photo = analogRead(A0); `

w loop() zatrzymuje mi resztę Smogomierza - nie generuje się web. To zapewne jakiś trywialny błąd myślowy z mojej strony - po raz pierwszy siadam do tego środowiska. 

Będę wdzięczny za podpowiedź.

Jednocześnie, po uruchomieniu chętnie podeślę jakiś howto, być może ktoś będzie miał podobną potrzebę.

bfaliszek, 2020-01-12:
Kod w loop() działa w Smogomierzu tak, że ciągle sprawdza czy jest jakiś klient dla serwera www. Co kilkadziesiąt sekund wykonuje jakąś inną operację. Jeśli doda się coś do samego loop(), to najprawdopodobniej będzie blokować obsługę www. 
Wydaje mi się, że najprościej będzie Ci dodać wykonywanie pomiarów do takeTHPMeasurements(). Będą one wykonywane wtedy kiedy pomiary temperatury/wilgotności/ciśnienia i nie będą blokować działania serwera www.
https://github.com/hackerspace-silesia/Smogomierz/blob/master/Smogomierz.ino#L845

jakubcichocki, 2020-01-15:
Super, działa. DziękI!

Nie jest to wielka zmiana, ale za +1 zł mamy badaną dodatkową wartość, być może dla kogoś użyteczną, stworzyć jakiś krótki opis?

bfaliszek, 2020-01-16:
Możesz podesłać swój kod, to dodam go do oprogramowania Smogomierzy.

skubi23t, 2020-01-20:
Jak wykorzystujesz ten fotorezystor?

jakubcichocki, 2020-01-20:
> Jak wykorzystujesz ten fotorezystor?

Do sterowania oświetleniem i roletami.

jakubcichocki, 2020-03-24:
Wrzuciłem swoje drobne zmiany na forku https://github.com/jakubcichocki/Smogomierz/commit/5b52dab09740dc440892dc7c1be84344461c6ba8

Jednakowo daleko im do ideału, obecność czujnika jest niesterowana, a tłumaczeń brak. Dodatkowa zmiana wystawiona jest do api json i do influxa. Chciałem to dopracować jednak czasu brak a na moje potrzeby wystarczyło, więc póki co wrzucam tak. Być może komuś się przyda, lub rozbuduje bardziej.

CoolTechProjects, 2021-04-12:
Czy w ramach rozwoju pojawi się także możliwość podłączenia dodatkowego czujnika (co2) z wysyłką danych na aqi oraz inne portale? Czujnik np. https://botland.com.pl/czujniki-czystosci-powietrza/19025-ccs811-czujnik-czystosci-powietrza-eco2-tvoc-i2c-dfrobot-sen0339.html

bfaliszek, 2021-04-12:
Z CCS811 mam jeden spory problem. Pomiary są tam mocno uzależnione od dobrej kalibracji, którą musi zabić użytkownik, bo każdy CCS811 różni się od siebie. IMO to zbyt duża loteria. Plus pomiar CO2 na zewnątrz nie ma specjalnie sensu. CO2 mierzysz w zamkniętych pomieszczeniach, żeby sprawdzić czy wentylacja jest ok. Smogomierz/Smogly z założenia służy do pomiarów zanieczyszczeń na zewnątrz.

CoolTechProjects, 2021-04-13:
dzięki za wyjaśnienia!