hackerspace-silesia/Smogomierz #70: Wyświetlanie wykresów z serwisu AirMonitor - restują się ustawienia thingspeaka

XsPolar, 2020-01-14:
Po zmianie parametru Wyświetlanie wykresów z serwisu AirMonitor: NIE  - resetują się ustawienia thingspeaka, wysokość i ustawienia dla luftdaten.

bfaliszek, 2020-01-16:
Jaka wersja oprogramowania? 2.7.2? Smogomierz z ESP8266 i PMS7003? Właśnie sprawdziłem na takim u siebie i wszystko zadziałało. Wyłączenie wykresów AirMonitora, nie wpłynęło na ustawienia Thingspeaka. 

Jedyne co przychodzi mi na myśl, to opcja z błędnym wczytaniem strony z konfiguracją. Jeśli na dole są teksty w "{}" i brakuje tłumaczeń, to przy zapisaniu, czasami przywraca ustawienia domyślne. Spowodowane jest to brakami pamięci w ESP8266. Tylko to dzieje się niezależnie od tego jakie zmieni się ustawienia. Jeśli jest za mało pamięci, to się nie zapiszą i dla bezpieczeństwa przywracane są te domyślne.

XsPolar, 2020-01-16:
Wersja 2.7.2 Chyba masz rację z tym brakiem pamięci, bo jest problem czasem reloadem strony konfiguracji po zapisaniu zmian. Ale tylko po zmianie tego parametru resetuje mi ustawienia. Pozostałe zmiany nie czyszczą danych do thinspeaka.

skubi23t, 2020-01-20:
U mnie też nie zawsze chce się wczytać strona konfiguracji tylko wyskakuję zmienne. Pewnie to ten brak pamięci..

bfaliszek, 2020-01-23:
Jeśli macie chwilę, to możecie przetestować prerelease 2.7.4(ESP8266 z PMSx003). Podzieliłem stronę Config na dwie mniejsze. Powinny się poprawnie wczytywać i dzięki temu nie powinno być już problemów ze znikaniem ustawień. https://github.com/hackerspace-silesia/Smogomierz/blob/dev/firmware/ESP8266/prerelease_ESP8266_Smogomierz_2.7.4_PMS_build_23.01.2020.bin

skubi23t, 2020-01-24:
Jest lepiej, ale i tak potrafi zrobić psikusa
[image]

diggahpl, 2020-02-02:
prerelease_ESP8266_Smogomierz_2.7.4_PMS_build_2.02.2020 - poprawiło wprowadzanie konfiguracji. Nie zaobserwowałem już pojawiania się {}.

skubi23t, 2020-02-02:
Fakt. U mnie to pojawiło się tylko raz niedługo po wygraniu.

bfaliszek, 2020-02-02:
W prerelease z lutego mocno zwiększyłem ilość pamięci przeznaczonej na config. O prawie 1/2. W dodatku zoptymalizowałem użycie pamięci, więc powinno być lepiej jeśli chodzi o problemy z {} czy znikającymi danymi z konfiguracji. Przypuszczam, że to było problemem przy wyświetlaniu wykresów z AirMonitor.

bfaliszek, 2020-02-06:
https://github.com/hackerspace-silesia/Smogomierz/releases/tag/v2.7.5

closed 2020-02-06 by bfaliszek