hackerspace-silesia/Smogomierz #73: Odczyt na żądanie

szczemp, 2020-01-18:
Zrobiłem sobie coś podobnego na bazie nidecmu v2 i pms7003. Chciałbym na stronie prezentować wyniki pomiarów. Zrobiłem prosty serwer i jako tako działa. To znaczy pokazuje mi odczyt, ale nie zawsze. Przy odświeżaniu strony potrafi pokazywać się przez wiele odświeżeń. A czasem po kilku poprawnych  odczytach prze kilka kolejnych nie odczytuje, po czym znowu wraca działanie.
W funkcji zwracającej dane na www mam taki kod:

`String json = "{\"smog\": {\"status\": \"0\"}}";
  while (PMS_Serial.available()) {
    PMS_Serial.read();
  }
  pms.requestRead();
  
  if (pms.readUntil(data)) {    
    json = "{\"smog\": {\"status\": \"1\", \"PM10\": "+ String((int)data.PM_AE_UG_1_0)+ ", \"PM25\": "+ String((int)data.PM_AE_UG_2_5)+ ", \"PM100\": "+ String((int)data.PM_AE_UG_10_0)+ "}}";
  }

  server.send(200, "text/html", json);`
Możecie pomóc?

bfaliszek, 2020-01-18:
PMS_Serial.available() zawsze działa, tylko nie zawsze podaje dane. Dane pojawiają się, wtedy kiedy PMS je wysyła. 

Jak często wykonujesz ten kod?

szczemp, 2020-01-18:
Na nodemcu jest serwer i pod linkiem 192.168.1.100/stan wyświetla dane odczytane z pms7003 w jsonie.
Zatem dane mają się pokazywać przy każdym odświeżeniu strony.
Skoro dane są dostępne dopiero jak czujnik je wyśle, to jak zrobić, żeby przy otwieraniu strony poczekać na te dane i dopiero potem je wyświetlić?

szczemp, 2020-01-19:
Trochę pokombinowałem i zrobiłem sowie workaround.
Najpierw według przykładu Basic zapisuję odczyty do zmienny, a potem te zmienne zwracam w jsonie przez wywołanie linku 192.168.1.100/stan. Ale tu czujnik cały czas wysyła dane do nodemcu, a ten zapisuje je do zmiennych. Kupa niepotrzebnego zapisywania.
Potem powstała druga wersja. Wszystko odbywa się w funkcji wywoływanej podczas otwarcia linku 192.168.1.100/stan. Po uruchomieniu funkcji startuje pętla nieskończona while i tam czeka na dane z czujnika. Jeśli nadejdą, to są zapisywane do jsona, pętla jest przerywana a jsona zwracany do klienta. Mniej więcej wygląda tak:
`String json = "{\"smog\": {\"status\": \"0\"}}";
    while (true) {
      //Serial.println("Próba");
      if (pms.read(data)) {
        pm10 = data.PM_AE_UG_1_0;
        //Serial.print("PM 1.0 (ug/m3): ");
        //Serial.println(pm10);
    
        pm25 = data.PM_AE_UG_2_5;
        //Serial.print("PM 2.5 (ug/m3): ");
        //Serial.println(pm25);
    
        pm100 = data.PM_AE_UG_10_0;
        //Serial.print("PM 10.0 (ug/m3): ");
        //Serial.println(pm100);
        
        //Serial.println();
        json = "{\"smog\": {\"status\": \"1\", \"PM10\": "+ String((int)pm10)+ ", \"PM25\": "+ String((int)pm25)+ ", \"PM100\": "+ String((int)pm100)+ "}}";
        break;
      }
    }
    server.send(200, "text/html", json);`

closed 2020-01-23 by bfaliszek