hackerspace-silesia/Smogomierz #74: Aqi zawyżona wartość ciśnienia

skubi23t, 2020-01-21:
Hej,
Skonfigurowałem wczoraj aqi
https://sqb.aqi.eco/dolna4/graphs

Wyniki pomiaru ciśnienia sa mocno zawyżone:
https://thingspeak.com/channels/671870/

To takie pierwsze spostrzeżenie, być może jest coś jeszcze czego na razie nie zauważyłem

krupniok200, 2020-01-21:
Potwierdzam, na stronie www 1038 a w aqi 1074

bfaliszek, 2020-01-23:
Dziwna sprawa. Wydaje mi się, że problem może leżeć po stronie aqi.eco - @trekawek. Gdy się zaloguje i w Lokacje wybierze swoje urządzenia, tam jest dostęp do danych prosto z przesłanego JSONa(tabelka "Otrzymane JSONy" i na dole jest przycisk "Pokaż wszystkie"). 

Mam np. u siebie `{"value_type":"BME280_pressure","value":"103547.05"}]}`, czyli powinno mi wyświetlać 1037 albo 1038 hPa, w zależności od tego jak zaokrągli. Tymczasem wyświetla mi "pressure 1065.9179541846" czyli 1066 hPa.

trekawek, 2020-01-23:
Na stronie wyświetla się ciśnienie znormalizowane do poziomu morza (takie jak w prognozie pogody). Do jego obliczenia używana jest wysokość ustawiana w konfiguracji czujnika na aqi.eco. Wpisanie "0" w pole wysokości wyłącza normalizację:

https://github.com/trekawek/air-quality-info/blob/c6a72b3a6ee195c2597f96b6640b725e3c9bee79/src/htdocs/model/record_model.php#L357-L373

skubi23t, 2020-01-23:
To trzeba raczej wpisać 0, bo przy moich 201m npm w dzień gdzie w Polsce najwyższe cisnienie było podobno 1045-1055 około to ja miałem na wykresach ~1070 co raczej nie jest wiarygodne.

trekawek, 2020-01-23:
@skubi23t - podeślesz linka do wykresu?

skubi23t, 2020-01-23:
W pierwszym poście masz dwa urle z mojego czujnika :)

trekawek, 2020-01-23:
Racja, dzięki. Nie zmieniaj na razie wysokości, przyjrzę się temu.

bfaliszek, 2020-01-23:
Smogomierz robi to sam z siebie. W konfiguracji ustawia się wysokość. Wartość wysyłana do aqi.eco jest już znormalizowana.

> Na stronie wyświetla się ciśnienie znormalizowane do poziomu morza (takie jak w prognozie pogody). Do jego obliczenia używana jest wysokość ustawiana w konfiguracji czujnika na aqi.eco. Wpisanie "0" [...]

trekawek, 2020-01-23:
W takim razie póki co proponuję wpisywać "0" w pole wysokości. Popracuję nad automatycznym wykrywaniem Smogomierza w aqi.eco, aby w przyszłości nie normalizować ciśnienia otrzymanego z tego typu urządzeń.

closed 2021-11-16 by bfaliszek