hackerspace-silesia/Smogomierz #75: Praca na zewnątrz i zasilanie z 18650

Riventros, 2020-01-25:
Witam!

Czy smogomierz byłby w stanie bezpiecznie pracować na zewnątrz? Obudowa ma IP67, chyba wywietrzniki byłby problemem? Gdyby w miejscu samego mUSB zastosować moduł ładowarki ogniw, a za ogniwem przetwornice step-up na 5V to miałoby to rację bytu? Jak wygląda pobór prądu całego układu. Czy można by zoptymalizować kod pod kątem energooszczędności np. pomiar co 15 minut, i tryb czuwania? 

Pozdrawiam

bfaliszek, 2020-01-25:
Można włączyć w ustawieniach Deepsleep i wybudzanie urządzenia co kilka-kilkanaście minut. 
Jeśli chodzi o obudowę to bez problemu urządzenie może działać na zewnątrz. Sam 5 Smogomierzy, każdy jest w takiej obudowie i jest zasilany przez USB. Najstarszy działa od 3 lat. Wszystkie są na zewnątrz.

Riventros, 2020-01-25:
Jak wygląda zasilanie z USB na zewnątrz?

bfaliszek, 2020-01-25:
No normalnie, mam podłączony kabel do Smogomierza. Chyba nie za bardzo rozumiem o co pytasz.

Riventros, 2020-01-25:
Ale jak pociągnięty jest kabel na zewnątrz?

bfaliszek, 2020-01-25:
najczęściej przekładam go przez okno i zamykam. Da się tak zamknąć okno bez problemu, a kabel nadal działa. Najczęściej tak przerzucam kable USB na zewnątrz.

Riventros, 2020-01-25:
A czy obudowa jest wystawiona na działanie np. deszczu?

bfaliszek, 2020-01-25:
Jasne. Jak już pisałem, przez 3 lata nie miałem problemów z obudową. Tak szczerze to najbardziej problematyczne są kabelki w środki. Czasami trzeba je wymienić po roku, a czasami działają trzy bez problemów

Riventros, 2020-01-26:
A z doświadczenia, który czujnik najlepiej zastosować?

ookris, 2020-01-26:
> Ale jak pociągnięty jest kabel na zewnątrz?

Tak jak pisał Błażej kabel zwyczajnie można "przyciąć" oknem. Polecam do tego kable typu "flat", które bez problemu można tak robić. 

https://www.trust.com/en/product/20135-flat-micro-usb-cable-1m-black

bfaliszek, 2020-01-26:
Jeśli chodzi o czujnik to polecam Sensirion SPS30 i do tego ESP32. No i BME280 do pomiarów temperatury, wilgotności i ciśnienia.

Kriztower, 2020-02-07:
[image]
U mnie tak to wygląda.

jankes46, 2020-11-30:
czy jest jakaś możliwość dołożenia panelu słonecznego ?

closed 2021-11-16 by bfaliszek