hackerspace-silesia/Smogomierz #76: Brak połączenia z influxdb

fibeer, 2020-01-30:
W konfiguracji dodałem swój serwer z influxdb, jednak za nic smogomierz nie wysyła żadnych danych, w ogóle nie próbuje się łączyć (sprawdzane tcpdumpem). Często też są błędy w konfiguracji, w sensie, że znika część konfiguracji, albo nie jest widoczna. Jednakże największym problemem dla mnie jest to, że za nic nie mogę zmusić smogomierza do wysyłania danych do influxa

bfaliszek, 2020-01-30:
Kończę właśnie nową wersję oprogramowania, która ma wyeliminować problemy ze stroną konfiguracyjną.
Jeśli masz chwilę, to możesz sprawdzić czy na 2.7.2 działa InfluxDB. Wiem, że ostatnio były jakieś zmiany w obsłudze ich baz danych.

fibeer, 2020-01-30:
Dzięki za szybki odzew. Obecnie mam wersję 2.7.3, czyli spróbować wgrać wcześniejszą wersję? Wiesz, tutaj widać, że smogomierz w ogóle nie wysyła jakichkolwiek pakietów do bazy, krótko mówiąc w ogóle nie próbuje się łączyć.

fibeer, 2020-01-30:
Wiesz, co od razu po wgraniu softu 2.7.2 smogomierz zaczął wysyłać dane do influxa. Także wydaję się działać w tej wersji, a mieć problemy z wersją 2.7.3.
Spróbuję jeszcze raz wgrać wersję 2.7.3

fibeer, 2020-01-30:
Niestety, jak tylko wgrałem wersję 2.7.3 to od razu przestał wysyłać dane. Wycofałem do 2.7.2 i od razu zaczął nadawać od influxa, także jest jakiś babol w 2.7.3. Może pamięci za mało? Mam układ ESP8266 - NodeMCU V3

bfaliszek, 2020-01-31:
OK, dzięki za info, czyli to tylko problem w wersji 2.7.3. Postaram się to jeszcze dokładnie sprawdzić i naprawić przy kolejnej aktualizacji.

ookris, 2020-02-01:
Potwierdzam, u mnie jest dokładnie tak samo. Dane do połączenia z serwerem poprawne a mimo wszystko nic nie jest wysyłane.

diggahpl, 2020-02-02:
prerelease_ESP8266_Smogomierz_2.7.4_PMS_build_2.02.2020 - dane też nie idą do Influxa ;(

bfaliszek, 2020-02-02:
Czyli chyba muszę przerobić całą obsługę Influx'a. W zasadzie to tylko to mi zostało i wizualne poprawni na stronie z zaawansowaną konfiguracja MQTT. Czyli w przyszłym tygodniu będzie nowe oprogramowanie 😁

fibeer, 2020-02-02:
Super, trzymam kciuki :)!

bfaliszek, 2020-02-03:
Jakie macie wersje InfluxDB(ocenie, stabilna jest 1.7.9)?
Od wersji 2(obecnie beta), zmieni się sposób w jaki autoryzuje się klienta. Nie wystarczy już login i hasło. Potrzebny będzie też np. token.

ookris, 2020-02-04:
Ja mam 1.7.9

diggahpl, 2020-02-04:
rówież  1.7.9.

bfaliszek, 2020-02-04:
Dobra, to na szybko, w tej wersji działa InfluxDB(testowałem z 1.7.9) z loginem i hasłem do bazy danych. Nie testowałem na bazie bez loginu(dodam wieczorem, sprawdzenie czy w ustawieniach jest podany login i hasło, jeśli nie będzie, to inaczej będę formatował adres do wysyłki danych). 
https://github.com/hackerspace-silesia/Smogomierz/blob/dev/firmware/ESP8266/prerelease_ESP8266_Smogomierz_2.7.4_PMS_build_4.02.2020.bin
Chcę jeszcze dodać sobie wstępnie w kodzie zmienną z wersją InfluxDB(1 albo 2). Dodałbym w konfiguracji możliwość wyboru. Przy wybraniu 2, pojawi się więcej okienek na dane(token itp.). Dodam to pewnie dziś wieczorem, albo w czwartek. Ogólnie, to chciałbym w czwartek puścić oficjalnie aktualizację z tymi wszystkimi poprawkami.

diggahpl, 2020-02-04:
> Dobra, naprawiłem obsługę InfluxDB(testowałem z 1.7.9) z loginem i hasłem do bazy danych. Nie testowałem na bazie bez loginu. Jeśli ktoś taką ma, to będę wdzięczny za info czy działa.

Testowane z auth. i działa :) dziekuję !

// raz zaobserwowałem {} przy auth passwd w konfigu.

bfaliszek, 2020-02-06:
https://github.com/hackerspace-silesia/Smogomierz/releases/tag/v2.7.5

closed 2020-02-06 by bfaliszek