hackerspace-silesia/Smogomierz #78: Zawieszający się ESP32

ookris, 2020-02-08:
Hej, przeszedłem właśnie na ESP32 i po małych problemach z podłączeniem kabelków mam kolejny. 
Najnowsze oprogramowanie po wykonaniu pierwszych pomiarów PM po prostu się zawiesza. Restart nic nie daje gdyż ponownie po wykonaniu pierwszych pomiarów PM wszystko staje. 

Log z Ardiunio:

`09:01:17.824 -> ets Jun  8 2016 00:22:57
09:01:17.824 -> 
09:01:17.824 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
09:01:17.824 -> configsip: 0, SPIWP:0xee
09:01:17.824 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
09:01:17.858 -> mode:DIO, clock div:1
09:01:17.858 -> load:0x3fff0018,len:4
09:01:17.858 -> load:0x3fff001c,len:1216
09:01:17.858 -> ho 0 tail 12 room 4
09:01:17.858 -> load:0x40078000,len:9720
09:01:17.858 -> ho 0 tail 12 room 4
09:01:17.858 -> load:0x40080400,len:6352
09:01:17.858 -> entry 0x400806b8
09:01:18.249 -> Mounting FS...
09:01:18.249 -> Loaded DEVICENAME_AUTO: 1
09:01:18.283 -> Loaded DEVICENAME: smogomierz1
09:01:18.283 -> Loaded LANGUAGE: polish
09:01:18.283 -> Loaded SELECTED_LANGUAGE: 2
09:01:18.283 -> Loaded THP_MODEL: BME280
09:01:18.283 -> Loaded DUST_MODEL: PMS7003
09:01:18.283 -> Loaded DISPLAY_PM1: 1
09:01:18.283 -> Loaded FREQUENTMEASUREMENT: 0
09:01:18.283 -> Loaded DUST_TIME: 5
09:01:18.283 -> Loaded NUMBEROFMEASUREMENTS: 10
09:01:18.283 -> Loaded LUFTDATEN_ON: 0
09:01:18.283 -> Loaded SMOGLIST_ON: 0
09:01:18.283 -> Loaded AIRMONITOR_ON: 0
09:01:18.283 -> Loaded AIRMONITOR_GRAPH_ON: 0
09:01:18.283 -> Loaded LATITUDE: 51.261981
09:01:18.316 -> Loaded LONGITUDE: 15.580108
09:01:18.316 -> Loaded MYALTITUDE: 182
09:01:18.316 -> Loaded THINGSPEAK_ON: 0
09:01:18.316 -> Loaded THINGSPEAK_GRAPH_ON: 0
09:01:18.316 -> Loaded THINGSPEAK_API_KEY: API-KEY
09:01:18.316 -> Loaded THINGSPEAK_CHANNEL_ID: 0
09:01:18.316 -> Loaded THINGSPEAK_READ_API_KEY: READ_API_KEY
09:01:18.316 -> Loaded INFLUXDB_ON: 1
09:01:18.316 -> Loaded INFLUXDB_VERSION: 1
09:01:18.316 -> Loaded INFLUXDB_HOST: db.bolecair.info
09:01:18.316 -> Loaded INFLUXDB_PORT: 8086
09:01:18.316 -> Loaded INFLUXDB_DATABASE: xxxxxxxxx
09:01:18.316 -> Loaded INFLUXDB_USER: xxxxxxxx
09:01:18.351 -> Loaded INFLUXDB_PASSWORD: xxxxxxxxx
09:01:18.351 -> Loaded INFLUXDB_ORG: 
09:01:18.351 -> Loaded INFLUXDB_BUCKET: 
09:01:18.351 -> Loaded INFLUXDB_TOKEN: 
09:01:18.351 -> Loaded MQTT_ON: 0
09:01:18.351 -> Loaded MQTT_HOST: MQTT_ADRESS
09:01:18.351 -> Loaded MQTT_PORT: 1883
09:01:18.351 -> Loaded MQTT_USER: username
09:01:18.351 -> Loaded MQTT_PASSWORD: password
09:01:18.351 -> Loaded MQTT_IP_IN_TOPIC: 0
09:01:18.351 -> Loaded MQTT_DEVICENAME_IN_TOPIC: 0
09:01:18.351 -> Loaded MQTT_TOPIC_TEMP: MQTT_TOPIC_TEMP
09:01:18.384 -> Loaded MQTT_TOPIC_HUMI: MQTT_TOPIC_HUMI
09:01:18.384 -> Loaded MQTT_TOPIC_PRESS: MQTT_TOPIC_PRESS
09:01:18.384 -> Loaded MQTT_TOPIC_PM1: MQTT_TOPIC_PM1
09:01:18.384 -> Loaded MQTT_TOPIC_PM25: MQTT_TOPIC_PM25
09:01:18.384 -> Loaded MQTT_TOPIC_PM10: MQTT_TOPIC_PM10
09:01:18.384 -> Loaded MQTT_TOPIC_AIRQUALITY: MQTT_TOPIC_AIRQUALITY
09:01:18.384 -> Loaded AQI_ECO_ON: 0
09:01:18.384 -> Loaded AQI_ECO_HOST: ???.aqi.eco
09:01:18.384 -> Loaded AQI_ECO_PATH: /u/???
09:01:18.384 -> Loaded SENDING_FREQUENCY: 5
09:01:18.384 -> Loaded SENDING_DB_FREQUENCY: 5
09:01:18.422 -> Loaded DEEPSLEEP_ON: 0
09:01:18.422 -> Loaded DEBUG: 1
09:01:18.422 -> Loaded AUTOUPDATE_ON: 0
09:01:18.422 -> Loaded CONFIG_AUTH: 0
09:01:18.422 -> Loaded CONFIG_USERNAME: admin
09:01:18.422 -> Loaded CONFIG_PASSWORD: password
09:01:18.422 -> Loaded MODEL: white
09:01:18.422 -> Loaded PMSENSORVERSION: PMS
09:01:18.422 -> Loaded SOFTWAREVERSION: 2.7.5 build 17:28:36 Feb  6 2020
09:01:18.422 -> 
09:01:18.422 -> 
09:01:18.422 -> Config loaded
09:01:18.422 -> Polish Translation!
09:01:18.422 -> 
09:01:19.336 -> Device name: Smogly-EC4F5FC
09:01:19.336 -> *WM: [1] AutoConnect 
09:01:19.408 -> *WM: [2] esp_wifi_set_country: US 
09:01:19.408 -> *WM: [2] ESP32 event handler enabled 
09:01:19.408 -> *WM: [2] Connecting as wifi client... 
09:01:19.408 -> *WM: [1] STA static IP:
09:01:19.408 -> *WM: [2] setSTAConfig static ip not set 
09:01:19.408 -> *WM: [3] WIFI station disconnect 
09:01:19.408 -> *WM: [1] Connecting to SAVED AP: Linksys2.4_1900ACS
09:01:19.408 -> *WM: [3] Using Password: xxxxxxxxxxxxxx
09:01:19.408 -> *WM: [3] WiFi station enable 
09:01:19.443 -> *WM: [1] connectTimeout not set, ESP waitForConnectResult... 
09:01:20.436 -> *WM: [2] Connection result: WL_CONNECTED
09:01:20.436 -> *WM: [3] lastconxresult: WL_CONNECTED
09:01:20.436 -> *WM: [1] AutoConnect: SUCCESS 
09:01:20.436 -> *WM: [1] STA IP Address: 10.0.0.63
09:01:20.436 -> connected...yeey :)
09:01:20.688 -> Waiting for NTP time sync: .
09:01:21.177 -> Current time: Sat Feb  8 08:01:21 2020
09:01:22.118 -> 
09:01:22.118 -> SERVER SOFTWARE VERSION: 2.7.5
09:01:22.118 -> CURRENT SOFTWARE VERSION: 2.7.5
09:01:22.118 -> 
09:01:22.118 -> You have the current version of the firmware!
09:01:22.118 -> 
09:01:22.118 -> HTTPServer ready! http://Smogly-EC4F5FC.local/
09:06:23.004 -> 
09:06:23.004 -> Turning ON PM sensor...
09:06:28.430 -> 
09:06:28.430 -> PM measurement number: 0
09:06:28.430 -> Value of PM1: 22
09:06:28.430 -> Value of PM2.5: 35
09:06:28.430 -> Value of PM10: 38
09:06:30.436 -> 
09:06:30.436 -> PM measurement number: 1
09:06:30.436 -> Value of PM1: 27
09:06:30.436 -> Value of PM2.5: 43
09:06:30.436 -> Value of PM10: 58
09:06:32.250 -> 
09:06:32.250 -> PM measurement number: 2
09:06:32.250 -> Value of PM1: 28
09:06:32.250 -> Value of PM2.5: 44
09:06:32.250 -> Value of PM10: 59
09:06:34.227 -> 
09:06:34.227 -> PM measurement number: 3
09:06:34.227 -> Value of PM1: 29
09:06:34.227 -> Value of PM2.5: 46
09:06:34.227 -> Value of PM10: 59
09:06:36.244 -> 
09:06:36.244 -> PM measurement number: 4
09:06:36.244 -> Value of PM1: 30
09:06:36.244 -> Value of PM2.5: 47
09:06:36.244 -> Value of PM10: 62
09:06:38.237 -> 
09:06:38.237 -> PM measurement number: 5
09:06:38.237 -> Value of PM1: 31
09:06:38.237 -> Value of PM2.5: 48
09:06:38.237 -> Value of PM10: 62
09:06:40.248 -> 
09:06:40.248 -> PM measurement number: 6
09:06:40.248 -> Value of PM1: 31
09:06:40.248 -> Value of PM2.5: 48
09:06:40.248 -> Value of PM10: 62
09:06:42.232 -> 
09:06:42.232 -> PM measurement number: 7
09:06:42.232 -> Value of PM1: 37
09:06:42.232 -> Value of PM2.5: 59
09:06:42.232 -> Value of PM10: 77
09:06:44.228 -> 
09:06:44.228 -> PM measurement number: 8
09:06:44.228 -> Value of PM1: 36
09:06:44.228 -> Value of PM2.5: 59
09:06:44.228 -> Value of PM10: 80
09:06:46.252 -> 
09:06:46.252 -> PM measurement number: 9
09:06:46.252 -> Value of PM1: 36
09:06:46.252 -> Value of PM2.5: 60
09:06:46.252 -> Value of PM10: 80
09:06:46.252 -> 
09:06:46.252 -> Average PM1: 30
09:06:46.252 -> Average PM2.5: 48
09:06:46.252 -> Average PM10: 63
09:06:46.252 -> Turning OFF PM sensor...
09:06:46.252 -> Measurements from BME280!
09:06:46.252 -> 
09:06:46.252 -> 
09:06:46.252 -> Measurements from PMSx003!
09:06:46.252 -> 
`

bfaliszek, 2020-02-09:
W tych logach jest wszystko OK. Przecież pomiary PM nie są wykonywane ciągle. Po uśpieniu PMS, powinno minąć trochę czasu do ponownego jego włączenia i nowej serii pomiarów. Nie ma w logach nic więcej?
Dodatkowe pytanie, użyłeś gotowego .bin czy samemu skompilowałeś kod?

ookris, 2020-02-09:
Tak wiem, problem z tym, że po pojawieniu się Measurements from PMSx003! nic się nie dzieje nie ważne jak długo bym czekał. Nie ważne czy mam ustawione pomiary co 5 minut czy co minutę. Za każdym razem smogomierz zawieszał się w tym samym momencie. 

Testowałem na dwóch płytach i za każdym razem to samo. 
Moje ESP: [ESP-32 WROOM WiFi+Bluetooth ESP32 NodeMCU Arduino](https://allegro.pl/oferta/esp-32-wroom-wifi-bluetooth-esp32-nodemcu-arduino-8644617352)

bfaliszek, 2021-11-16:
Nie wiem czy problem jeszcze występuje, ale w ostatnich comitach w branchu DEV skupiłem się głownie na ESP32 i teraz działa prawidłowo z PMS7003.