hackerspace-silesia/Smogomierz #79: nie chce skompilować wersji 2.7.5

krupniok200, 2020-02-09:
Dziś próbowałem skomplikować repo z wersji 2.7.5 i wyskakuje błąd podany w linku, niby za duża ilość bibliotek, ale ja tam nie potrafię się doliczyć, z programem jest wszystko ok, bo poprzednia kompilacja przechodzi bez problemów, biblioteki najnowsze

Błąd: [https://pastebin.com/Z1x9ZjJQ](url)

majstercar, 2020-02-09:
Mi pomogło dodanie ręcznie do Arduino bibliotek:ESPAsyncTCP i  ESPAsyncWebServer - Pozdrawiam

krupniok200, 2020-02-09:
zapodasz jak to zrobiłeś? bo w katalogach bibliotek tego nie widzi mi

bfaliszek, 2020-02-09:
ESPAsyncTCP - https://github.com/me-no-dev/ESPAsyncTCP
ESPAsyncWebServer - https://github.com/me-no-dev/ESPAsyncWebServer
Pobierasz .zip z biblioteką. W ArduinoIDE dodajesz te .zip z bibliotekami.

krupniok200, 2020-02-09:
dzięki, działa!

bfaliszek, 2020-02-09:
Dodałem info o tych bibliotekach do instrukcji - https://github.com/hackerspace-silesia/Smogomierz/blob/master/instrukcje/software.md#arduino---instalacja 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2020-02-09 by bfaliszek