hackerspace-silesia/Smogomierz #80: Informacja przez LED o stanie powietrza

dudus2000, 2020-02-09:
Witam.
Na wstepie przepraszam jezeli moje pytania sa niezbyt powazne, Jestem samoukiem i probuje sie zmierzyc z ta materia.
Nie moge sobie poradzic w dostosowaniu kodu do obslugi diody LED na obudowie smogomierza.
Dioda bedzie infomowala o stanie powietrza bez potrzeby odpalania strony.
Najpierw usuniete zostaly  zdefiniowane piny 12 13 i 14 z DHT22 i SHT1x (D5,D6,D7) w sumie to wszystkim wpisalem jeden nie uzywany pin 10

zdefiniowalem od nowa w/w piny
1sposob
#define ledR 14
#define ledR 12
#define ledR 13

2sposob
const int ledR = 14; //led red
const int ledG = 12; //led green
const int ledB = 13; //led blue
.............................
**nastepnie w :**
void setup() {
// dodany kod
  pinMode(ledR, OUTPUT);
  pinMode(ledG, OUTPUT);
  pinMode(ledB, OUTPUT);
.............................................
 **i w**
void loop() {
krysian();
   yield();
............................................
 **a na koncu funkcja**

void krysian() {
if (averagePM25 <= 10) {
       digitalWrite(ledG, HIGH);
       delay(200);
        digitalWrite(leG, LOW);
        delay(200);
      } else if (averagePM25 > 10 && averagePM25 <= 20) {
        digitalWrite(ledG, HIGH);
        digitalWrite(ledR, HIGH);
      } else if (averagePM25 > 20 && averagePM25 <= 25) {
        digitalWrite(ledB, HIGH);
        digitalWrite(ledR, HIGH);
      } else if (averagePM25 > 25 && averagePM25 <= 50) {
        digitalWrite(ledR, HIGH);
      } else if (averagePM25 > 50) {
        digitalWrite(ledR, HIGH);
        delay(500);
        digitalWrite(ledR, LOW);
        delay(500);
      } else {
        digitalWrite(ledR, HIGH);
        delay(200);
        digitalWrite(ledR, LOW);
        delay(200); 
      }
      yield();
  }

efekt:
ze wzgledu na brak podlaczonego czujnika pm mam averagePM25 = 0 - przynajmniej tak pokazuje na stronie ;).
dioda mruga przez jakies 10-20 sek i zatrzymuje sie swiecac 
roznie juz kombinowalem i wciaz mam efekt ze sie tak jakby zawieszalo "mruganie"

Przepraszam jezeli nie w tym dziale ale nie wiem gdziebym mogl zadac to pytanie

pozdrawiam
krystian

bfaliszek, 2020-02-09:
> delay(200); 
Raczej nie polecam używać delay. Delay blokuje wszystko inne i w czasie delay, ESP8266 nie robi nic, czyli nie wykonuje w tle pomiarów itp. Lepiej użyć millis(). Tutaj przykład jak zrobić pętlę, która przez 20 sekund, będzie wykonywać coś co jest wewnątrz niej, a później przejdzie dalej - https://github.com/hackerspace-silesia/Smogomierz/blob/master/Smogomierz.ino#L869-L876 
To najprawdopodobniej powoduje, że program zatrzymuje się na tym świeceniu i nie robi nic więcej.

Kolejna sprawa to delay(200); oznacza wstrzymanie pacy na 200ms. Jeśli wpiszesz delay(1000) to masz wstrzymanie na prawie sekundę. Takie szybkie zmiany mruganie może być problematyczne.

Jak rozumiem masz 3 diody, a nie jedną RGB, która zmienia kolory?W przypadku diody RGB(np. ws2812), nie musisz definiować trzech pinów, tylko na jednym masz wszystko. Plus są bardzo fajne biblioteki do ich obsługi, na przykład - https://github.com/Makuna/NeoPixelBus

Kolejna sprawa - w jaki sposób masz podłączone te trzy diody LED?

Kriztower, 2020-02-09:
Prędzej bym widział żeby inna płytka "odczytywała" pomiar że Smogly i wyświetlała status na LED.

Czujnik jest na zewnątrz, a wizualizacja led w środku 😀

bfaliszek, 2020-02-09:
> Prędzej bym widział żeby inna płytka "odczytywała" pomiar że Smogly i wyświetlała status na LED.
> 
> Czujnik jest na zewnątrz, a wizualizacja led w środku 😀

Aktualne dane pomiarowe są zawsze dostępne w formie JSON pod adresem IP_Twojego_SMogly/api możesz pobierać dane z tego JSONa i wizualizować za pomocą LED czy wyświetlać na ekranie.

dudus2000, 2020-02-09:
Czujnik (raczej swiatlo diody) chcialem wystawic na widok wszystkich w okolicy i sa tez ludzie starsi nie obslugujacy komputerow a telefon sluzy im tylko do rozmawiania
Chcialbym stworzyc mozliwosc korzystania przez nich z pomiarow czujnika bez używania jakichkolwiek sprzetow
fajnie by bylo docelowo wystawic duuuuza diode ktorej barwa byla odzwierciedleniem jakosci powietrza
co do millis prubuje go okielznac

majstercar, 2020-02-09:
Ale dostałem do myślenia mam w domu gdzieś paski ip65 na ws (od myfrendów) i tak wymyśliłem sobie dwa paski na obudowie  jeden PM10 drugi PM2.5

dudus2000, 2020-02-09:
Chyba z millis zadzialalo, narazie sie nie zawiesza, ale jak mrugnie niebieska dioda na esp8266 to widac ze troche sie wydluza przejscie  diody mrugajacej  w stan przeciwny

Kriztower, 2020-02-09:
Trochę zrobił się offtopic😀 ale pożyteczny
Podzielcie się panowie kodem swoich projektów dla takich nieogarow jak ja😀

dudus2000, 2020-02-09:
Oglaszam sukces :) dzieki waszej pomocy oczywiscie :)
Podaje swoja konfiguracje
NodeMCU v3, BME280, PMS7003

do pinow D5 D6 i D7 oraz 3V podlaczylem zwykla 4 pionowa diode rgb ktora bedzie pokazywala stan powietrza
kod zaraz podesle, tylko musze to jakos uporzadkowac , bo straszny burdel mi sie do niego wkradl

dudus2000, 2020-02-09:
zanieczyszczenie - kolorow

0-10 zielony
10-20 mruga zielony
20-25 zolty
25-50 czerwony
powyzej 50 mruga czerwony
[Smogomierz.ino.nodemcu.zip](https://github.com/hackerspace-silesia/Smogomierz/files/4177454/Smogomierz.ino.nodemcu.zip)

Kriztower, 2020-02-09:
Mieszkam na Śląsku, więc muszę zmienić progi, bo będę miał cały czas na czerwono 😋
Gratulacje! Dobra robota. Jak chińczyk wyzdrowieje (koronabadziew) to wyślę mi moje diody RGB. Miałem zrobić z nich sygnalizator świateł na przycisk dla dziecka, ale priorytety się dzisiaj zmieniły 😀 pozdrowienia z malowniczego Sosnowca 😀

ookris, 2020-02-10:
> Mieszkam na Śląsku, więc muszę zmienić progi, bo będę miał cały czas na czerwono 😋
To raczej niewiele da bo jakość powietrza i tak będzie zła. Są pewne standardy, których należy się trzymać nawet jeśli dioda będzie się prze 90% czasu świeciła na czerwono.

bfaliszek, 2020-02-11:
W naszym wypadku(w sensie Polski) mamy do wyboru dwa standardy:

+ Polski Indeks Jakości Powietrza: http://powietrze.katowice.wios.gov.pl/indeks-jakosci-powietrza

+ Europejski Indeks Jakości Powietrza: http://airindex.eea.europa.eu (legenda do mapki)

W zależności od tego czy ktoś woli mniej rygorystyczny(polski) czy bardziej rygorystyczny(europejski) indeks jakości powietrza, na takie kolory powinna świecić dioda(lub je zmieniać przy takich wartościach jak w którymś z tych indeksów). 

Osobiście zalecam europejski. To, że u nas jest tragicznie źle, nie oznacza, że mamy zrezygnować z wyższych standardów i do nich nie dążyć.

closed 2021-11-16 by bfaliszek