hackerspace-silesia/Smogomierz #81: Róznica miedzy odczytem temperatury i cisnienia z czujnika a wyswietlanym na aqi.eco

dudus2000, 2020-02-09:
Znacznie roznia sie dane z czujnika z tymi przedstawianymi na stronie
czy istnieje jakas mozliwosc zniwelowania tej roznicy
w zalacznikach pokazuje jak to u mnie wyglada
[image]
[image]

krupniok200, 2020-02-09:
było to już, ustaw na stronie wysokość 0 i będzie poprawnie

bfaliszek, 2020-02-09:
https://github.com/hackerspace-silesia/Smogomierz/issues/74#issuecomment-577683928

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2020-02-09 by bfaliszek