hackerspace-silesia/Smogomierz #82: Nawiązywanie połączenia z WiFi po zaniku napięcia

golytm, 2020-02-11:
Witam.

Zauważyłem, że po chwilowym zaniku napięcia, podczas którego resetuje się zarówno router WiFi jak i Smogomierz, nie dochodzi do ponownego połączenia z siecią WiFi. Z tego co udało mi się ustalić, wynika to z tego że router po prostu dłużej się podnosi i inicjalizuje sieć, niż Smogomierz. Innymi słowy Smogomierz próbuje połączyć się do sieci której jeszcze nie ma w powietrzu.
Sytuacja jest permanentna - za każdym razem przebieg jest ten sam. 
W kodzie jest metoda "WiFi_autoReconnect()" która wg mojego zrozumienia powinna podejmować próbę ponownego połączenia się z WiFi.
Pytanie czy to bug, czy może jednak intencjonalne działanie - np. do uruchamiania AP (nie sprawdziłem tego czy pojawia się AP).

Wojtek

bfaliszek, 2020-02-11:
W oprogramowaniu 2.7.5, jeśli urządzenie nie jest podłączone do WiFi, próbuje nawiązać ponownie połączenie ze skonfigurowaną wcześniej siecią co 5 minut.

closed 2020-02-11 by bfaliszek

radziejewicz, 2023-04-16:
@bfaliszek  na pewno? W wersji ESP8266_Smogomierz_2.8.1_PMS_build_27.11.2021 po resecie nie nawiązuje połączenia.