hackerspace-silesia/Smogomierz #83: Smogly 2.7.5 nie wysyła danych do InfluxDB

ookris, 2020-02-15:
Tak jak w tytule. Najnowsza wersja oprogramowania, zainstalowana na ESP8266 nie wysyła danych do bazy Influx v1.x. 
Wszystko działa ładnie, pomiary wyświetlają się na stronie, samo oprogramowanie po 30-40 minutach testowania wydaje się działać nieco bardziej responsywnie niż poprzednie wersje. 
Jutro jak znajdę czas podepnę płytkę do komputera i podrzucę log.

bfaliszek, 2020-02-16:
Masz influx w wersji 1.7.9? Tak jak pisałeś w tym wątku? Tutaj ktoś testował i było ok. Ja też sprawdzałem z 1.7.9 i działało poprawnie. Nie wiem jak przy bazie bez loginu i hasła, bo tego nie testowałem. Nie wiem też czy influxdb w wersji 2.0 działa poprawnie, ale tutaj pewnie coś się jeszcze zmieni przed pojawieniem się stabilnej wersji https://github.com/hackerspace-silesia/Smogomierz/issues/76#issuecomment-582011723

ookris, 2020-02-16:
OS: Ubuntu 16.04
InfluxDB: 1.7.10
Log w Arduino pokazuje, że niby dane do Influx są wysyłane ale Grafana już pokazuje zupełnie co innego.
Log: https://pastebin.com/5JWPSSvV
[image]