hackerspace-silesia/Smogomierz #84: ESP-Wroom-32 - podłączenia przewodów

ebuke, 2020-02-15:
Cześć,

Potrzebuję wsparcia, jak podłączyć przewody do budowy smogomierza, z modułemESP-Wroom-32? Kupiłem taki zgodnie z zaleceniami w instrukcji (https://github.com/hackerspace-silesia/Smogomierz/blob/master/instrukcje/hardware.md) i zacząłem składać, ale brakuje mi pinu GDN.   ESP32 ma 2 piny GND, a potrzeba 3 wg schematu. Będę wdzięczny, za podesłanie schematu, jak zbudować smogomierz w oparciu o moduł ESP32, albo wsparcie jak poradzić sobie z tym podłączeniem. Myślałem o zlutowaniu rodzielacza, tak jak wskazane jest dla zasilania 5V w instrukcji, ale wolę się upewnić czy to dobra droga.

krupniok200, 2020-02-15:
to jest ok droga, możesz zrobić osobną żyłę masową dla czujnika, a masę z czujnika temp/ciśnienia/wilg. do płytki

bfaliszek, 2020-02-16:
Możesz rozdzielić GND tak samo jak rozdzielone jest 5V. W zależności od producenta płytki z ESP, różnie są wyprowadzone piny. Czasami jest kilka GND, czasami jest jedna. Jeśli brakuje, to najprościej zrobić sobie taki rozdzielacz dla 5V i GND, bo tego najczęściej brakuje.

ebuke, 2020-02-23:
Wszystko działa, dzięki. Dla tych którzy kiedyś będą mieli podobne problemy, to opis podłączenia przewodów jest w komentarzu w kodzie.

closed 2020-02-23 by ebuke