hackerspace-silesia/Smogomierz #85: Nie zapisuje ustawień po wyborze czujnika zanieczyszczeń

krzaknet, 2020-02-15:
Wersja 2.7.5

Po wyborze i zapisaniu ustawień (czujnik zanieczyszczeń - Sensirion SPS30 ) nie zapamiętuje ustawień.

[image]

Testowane na ostatnim Safari oraz Chrome

bfaliszek, 2020-02-16:
Samemu kompilowałeś kod, czy wgrałeś gotowy .bin?

krzaknet, 2020-02-16:
Gotowy .bin

krzaknet, 2020-02-16:
Chwilowo udało mi sie obejsc problem, kompilujac samodzielnie .bin ze zmienionym defaultowym configiem.

bfaliszek, 2020-02-17:
A co w nim zmieniłeś?

bfaliszek, 2020-02-17:
Problem z zapisem konfiguracji wynika raczej z problemów z zapisem danych do pamięci urządzenia(SPIFFS) niż zmianami w konfiguracji. defaultConfig używany jest tylko jeśli nie można stworzyć/odczytać pliku config.json, który jest w SPIFFS.

leventehuszar, 2020-02-17:
Hi,
I have so this problem, too.
I installed sps30 sensor and when I would setup the dust sensor type on the interface site, the esp droped the sensors.
I used the original .bin. I probed the last and the previous version, too. Both firmware do it this.

krupniok200, 2020-02-18:
Można powiedzieć że też mam podobny problem, ja kompiluje sam już z spreparowanym configiem, po wgraniu .bin nie chce wgl robić pomiarów, lecz pomaga zmiana ustawienia (ja np włączam debuga, to jest pewnie w 100% że zadziała), w ostatniej kompilacji zmiana ustawień powoduje usunięcie i wyłączenie przesyłania do aqi, ponowne zapisanie ustawień z aqi kończy się fiaskiem

krzaknet, 2020-02-18:
@bfaliszek 

Zmieniałem defaultowy config (DUST_MODEL) po problemach z zapisywaniem ustawień konfiguracji z oficjalnej .bin

Problem dotyczy pola pm 2.5 sensor w interfejsie web - po wybraniu i zapisaniu konfiguracji zawsze wraca pole defaultowe "without sensor"

Problem ten np. nie dotyczy na przykład pola wyżej  - "Temp / Hum".

Wszystko robione wg. tutoriala
- czysty kontroler
- czyszczenie config podczas wgrywania .bin

Cofnąłem się jedną wersje wstecz, to samo.

krzaczek, 2020-02-18:
Jeżeli chodzi o AQI to mam ten sam problem.

Pomaga zmian tej lini https://github.com/hackerspace-silesia/Smogomierz/blob/a7b6773d4681662d4a2770091fb0b89111380419/src/spiffs.cpp#L349

na 

```
json["AQI_ECO_ON"] = String(AQI_ECO_ON);
```

Analogicznie jak jest `AQI_ECO_PATH`

bfaliszek, 2020-02-18:
@krzaczek 
dodałem sugerowane przez Ciebie zmiany do kodu. Są na razie w branchu dev - https://github.com/hackerspace-silesia/Smogomierz/tree/dev
pełna lista zmian - https://github.com/hackerspace-silesia/Smogomierz/commit/00bb837c4c56c85a915c67c1aea11ce91f1c94ab
W zasadzie to spisałem wszędzie typy zmiennych.

bfaliszek, 2020-02-20:
Sprawdźcie teraz, bo możliwe, że problem leżał gdzie indziej i przy zmianie miernika PM pobierały zły plik .bin z serwera.

michlimek, 2020-04-19:
Też natrafiłem na ten problem z zapisem ustawień AQI_ECO. Przy używaniu gotowej paczki wszystko jest ok, jednak przy samodzielnej kompilacji kodu nie. Znalazłem w pliku spiffs.cpp podwójny zapis ustawień. Usunięcie jednego u mnie pomogło.
AQI_ECO_ON = json["AQI_ECO_ON"];
  _safeCpy(AQI_ECO_HOST, json["AQI_ECO_HOST"], "host");
  _safeCpy(AQI_ECO_PATH, json["AQI_ECO_PATH"], "path");

bfaliszek, 2020-04-20:
Poprawiłem w branchu Dev